stanford-ppl/spatial #8: Accesses with Switch LCA Assigned to Same muxPort

mattfel1, 2018-05-04:
For example, both of these writes in NestedIf are getting placed on muxPort 0:

```
      val sram = SRAM[Int](3)
      if (in >= 42.to[Int]) {     // if (43 >= 42)
        if (in <= 43.to[Int]) {   // if (43 <= 43)
          sram(in - 41.to[Int]) = 10.to[Int] // sram(2) = 10
        }
      }
      else {
        if (in <= 2.to[Int]){
          sram(in) = 20.to[Int]
        }
      }
```

mattfel1, 2018-05-04:
I think I fixed this by making these changes.  Looking for things that this may have broken now

```
  def requireParallelPortAccess(a: Sym[_], b: Sym[_]): Boolean = {
    val lca = LCA(a,b)
    // TODO[2]: Arbitrary dataflow graph for children
    lca.isInnerPipe || lca.isInnerStream || lca.isParallel || lca.isSequential || lca.isSwitch
  }
```

and 

```
  protected def computePorts(groups: Set[Set[AccessMatrix]], bufPorts: Map[Sym[_],Option[Int]]): Map[AccessMatrix,Port] = {
        ...
        }.map{b => ports(b).muxPort + 1 }.maxOrElse(0)
        ...
  }
```

dkoeplin, 2018-05-04:
Sorry the documentation for requireParallelPortAccess was incomplete. The intended behavior is that it should return true if both accesses with be active concurrently AND are accessing the same buffer port. So Sequential and Switch LCAs should not be true here, because they will never be enabled concurrently. 

The other change looks correct.

I think the problem here actually stems from me misassigning muxPorts across sequentialized controllers. The original purpose of muxPort was for handling sequentialized accesses within a single controller, but it makes sense for it to also cover muxes across controllers.

dkoeplin, 2018-06-01:
General problem has been solved with the new muxPort / muxOfs metadata that the memory analysis sets. Any other problems with using the same muxPort across sequential accesses can be created as separate issues with this new implementation.

closed 2018-06-01 by dkoeplin