stanford-ppl/spatial #13: Counters Not Cleaned Up After FullUnroll

mattfel1, 2018-05-08:
Are they supposed to be removed?  Ex

```
Foreach(16 par 16) {_ => ... }
```

gives

```
              x143 = CounterNew(Const(0: class emul.FixedPoint),Const(16: class emul.FixedPoint),Const(1: class emul.FixedPoint),Const(16: class emul.FixedPoint))
               - Aliases: 0015: x7
               - Type: Counter[Fix[TRUE,_32,_0]]
               - SrcCtx: TwoDuplicatesSimple.scala:19:20
               - class argon.Effects: Unique
               - class spatial.data.ReadUses: ReadUses(Set())
               - class argon.Consumers: Consumers(Set(x144, x192, x304))
               - class spatial.data.ParentCtrl: ParentCtrl(Controller(x304,0))
               - class spatial.data.ParentBlk: ParentBlk(Controller(x304,0))
              x144 = CounterChainNew(List(x143))
               - Aliases: 0015: x9
               - Type: CounterChain
               - SrcCtx: TwoDuplicatesSimple.scala:19:27
               - class argon.Effects: Unique
               - class spatial.data.ReadUses: ReadUses(Set())
               - class argon.Consumers: Consumers(Set(x147, x192, x304))
               - class spatial.data.ParentCtrl: ParentCtrl(Controller(x304,0))
               - class spatial.data.CounterOwner: CounterOwner(x147)
               - class spatial.data.ParentBlk: ParentBlk(Controller(x304,0))
              x270 = UnitPipe(Set(b253),Block(Const((): class scala.runtime.BoxedUnit))) {
               - Aliases: 0023: x147, 0015: x12
               - Type: Void
               - SrcCtx: UnrollingBase.scala:106:61
               - class argon.Effects: (writes={x251})
               - class spatial.data.ReadUses: ReadUses(Set())
               - class argon.Consumers: Consumers(Set(x304))
               - class spatial.data.ControlLevel: InnerControl
               - class spatial.data.ParentCtrl: ParentCtrl(Controller(x304,0))
               - class spatial.data.ControlSchedule: ControlSchedule(Sequential)
               - class spatial.data.Children: Children(List())
               - class spatial.data.ParentBlk: ParentBlk(Controller(x304,0))
```

dkoeplin, 2018-05-21:
Unused counters / counterchains should be getting removed. There's a specific part of RegCleanup that does this now, but need to confirm that it's working for unrolled loops.

mattfel1, 2018-05-21:
Looks correct to me.  I changed `CounterRemovalOnUnroll` app because there was a store which will create a counter and mess up the check.

closed 2018-05-21 by mattfel1