stanford-ppl/spatial #18: Banking Calculation in DiagonalBanking is Wrong

mattfel1, 2018-05-18:
Probably already known, but the app DiagonalBanking is broken because it seems to be having trouble figuring out the banks for the read ports.  The node is `x603 = SRAMBankedRead(x298,Vector(List(x598), List(x598), List(x598)),Vector(x600, x601, x602),List(Set(b585, b586), Set(b585, b587), Set(b585, b588)),Vec[Fix[TRUE,_32,_0]])` which is a red flag because all three elements in the vector point to the same bank at different addresses

mattfel1, 2018-06-21:
I think this is the same bug but here is another app that I'm using right now to work on this

```
@test class MultilevelPar extends SpatialTest {
  override def runtimeArgs: Args = NoArgs

  def main(args: Array[String]): Unit = {
    val d = DRAM[Int](16,16)
    Accel {
      val s = SRAM[Int](16,16)
      Foreach(16 by 1 par 2, 16 by 1 par 4){(i,j) => s(i,j) = i+j}
      d store s
    }

    val result = getMatrix(d)
    val gold = (0::16, 0::16){(i,j) => i+j}
    printMatrix(result, "Result:")
    printMatrix(gold, "Wanted:")
    println(r"Pass? ${result == gold}")
    assert(result == gold)
  }

}
```

dkoeplin, 2018-06-28:
This appears to be fixed now thanks to your fixes to offset calc

closed 2018-06-28 by dkoeplin