stanford-ppl/spatial #25: Address Calculation for < full rank DRAM Wrong

mattfel1, 2018-05-21:
I think the offending code is the else branch of 

```
    val dramOffsets: Seq[I32] = dram.starts()
...
    if (counters.length > 1) {
      Stream.Foreach(counters.dropRight(1).map{ctr => ctr()}){ is =>
        val indices = is :+ 0.to[I32]

        val dramAddr = () => flatIndex((dramOffsets,indices,strides).zipped.map{case (ofs,i,s) => ofs + i*s }, dims)
        val localAddr = {i: I32 => is :+ i }
        if (isLoad) load(dramAddr, localAddr)
        else        store(dramAddr, localAddr)
      }
    }
    else {
      Stream {
        val dramAddr = () => flatIndex(dramOffsets, dims)
        val localAddr = {i: I32 => Seq(i) }
        if (isLoad) load(dramAddr, localAddr)
        else        store(dramAddr, localAddr)
      }
    }
```

It just takes the DRAM pointer for the request and I think it should be adding some to this in the case of loading one row out of a 2D mem but I need to look more closely

dkoeplin, 2018-05-21:
We should be filtering out counters here which are "Unit". Looks like I made a mistake when porting that part. Compare the old version of the expansion for DenseTransfer to this one, specifically around the line: 

`val counters: Seq[() => Counter[I32]] = lens.zip(pars).map{case (d,p) => () => Counter[I32](start = 0, end = d, par = p) }
`

mattfel1, 2018-05-22:
Added rawRankOf which gives a Seq(0 until true dims), which can be checked against rankOf which now gives a Seq(non-eliminated dims) and recover the appropriate address calculation.  Also added Alias3Dto2DTransfers test

closed 2018-05-22 by mattfel1