stanford-ppl/spatial #28: Controller children design is a bit wonky

dkoeplin, 2018-05-26:
Currently have children defined as:
  sym: `all controllers in all scopes`
  ctrl: `if the id is -1, all subcontrollers (id > -1), otherwise, all controllers in the given scope.`

This poses an immediate problem with MemReduce. See MultiWriteBuffer as an example:
```
MemReduce(...){
  val x = SRAM
  Foreach(...) { write(x) }
  Foreach(...) { write(x) }
  x
}{...}
```
creates a controller tree like:
 ```
MemReduce (-1)
  MemReduce (0)
    Foreach (-1)
      Foreach (0)
        { write }
    Foreach (-1)
      Foreach (0)
        {write}
  MemReduce (1)
    {read}
```
Currently calls the LCA the MemReduce(-1) and gives a buffer depth of 2.

Better interpretation would be:
```
MemReduce(-1)
  Foreach(-1)
    {write}
  Foreach(-1)
    {write}
  MemReduce(1)
    {read}
```

Required improvements for this:
  - Change name of children metadata to .rawChildren 
  - Add new definition of children on symbols which returns the same as Controller(s, -1)
  - Add concept of "transparent" (naming?) subcontroller which doesn't represent a true future stage
  - Redefine parent setter:  
    If the current scope is a "transparent" subcontroller, set parent to the master control (-1) instead.
  - Add "owner" setter for control: 
    Always set this to the exact subcontroller we're currently in.
  - Redefine controller / subcontroller children method: 
    If id = -1, list all subcontrollers but flatten transparent subcontrollers in place to their children
    Otherwise, list all children of the symbol which have an owner (not parent) that is this subcontroller

The purpose here is to maintain scoping information while also generally keeping the relationship that 
for all c in x.children, c.parent = x 
and for all symbols s with s.parent = x, x.children contains s
(This is needed to keep the tree traversal rules for controllers relatively sane).

Note that this rule won't hold for transparent subcontrollers - this plan effectively cuts them out of the controller tree hierarchy (they can be used as entry points, but you can never trace back to them)

@mattfel1 does this seem sane? If so I can go ahead and make these changes. You shouldn't see much impact at code generation time, as subcontrollers are only used prior to unrolling.

mattfel1, 2018-05-26:
This looks like it makes sense.  One question: if I understand, the MemReduce(0) became transparent/flattened away, does this make it tricky to handle the counterchains associated with controllers if these kinds of nodes disappear? Or is it just that you can handle the counterchains upon descending the tree and going back up doesn't matter? 

Another unrelated question just for my own understanding on how the buffering actually works right now in this case.

You say the LCA of the accesses would be MemReduce(-1), but aren't we supposed to crash today in that particular case if there is a pairwise LCA of sched Pipelined, and a triple-wise LCA higher up in the hierarchy that is Pipelined?  It would require a double-buffered double-buffer I think.  Or is there already some notion of "transparent" controllers that recognizes MemReduce(0) and MemReduce(1) being related and can resolve this?

dkoeplin, 2018-05-26:
That's right, the counterchains are associated through the ID, so we would have to use the "owner", not the parent, to identify the counterchains over a symbol. 

That's also correct, we were actually seeing an issue (ambiguous metapipes) in a couple apps due to this behavior (including in MultiWriteBuffer). I temporarily changed it to identify the LCA as control.master when looking for metapipelines, and that gives the correct metapipe without ambiguity, but also causes the above issue with incorrect depth.

dkoeplin, 2018-05-30:
Pending tests on hierarchy-fix branch

dkoeplin, 2018-06-02:
Fixed. We now have three hierarchies:
Scope - exact block, includes logical stage and block ID within that stage. Control iterators for access pattern analysis use this tree.
Ctrl - stage only, ignores "pseudostages" (final name for transparent subcontrollers). Most parent, child, ancestor operations use this tree.
Blk - linear ID for block. Usage analysis (in terms of register duplication, etc.) uses this tree.

closed 2018-06-02 by dkoeplin