stanford-ppl/spatial #31: Buffer Depth Wrong in NamedTest

mattfel1, 2018-05-29:
This may be happening in a few places, but specifically I am looking at `a_0` which has at least 2 things wrong.

1) The LCA is the accelscope so I think it should not be buffered
2) Its getting tagged as buffered with depth = 2, which even if it did have a metapipe lca, its buffer depth should be 4

dkoeplin, 2018-05-30:
Something weird is going on here with port calculation, looking into it tonight. 

Looks like the depth is 2 because `a` is used as MemReduce result and accumulator in a couple of places, but the ports are incorrect for this even then

dkoeplin, 2018-05-30:
Ok, this is actually a pretty interesting example. At the moment we have two relevant rules: 
- No more than one metapipe per instance
- No merging instances when at least one has "accumulating" writes (write + read in one controller) in a buffer

```
Instance #0
  <Banked>
  Depth:    2
  Accum:    Buffer
  Banking:  List(Dims {0}: Cyclic: N=1, B=1, alpha=<1>) <Flat>
  Pipeline: Controller(x101,-1)
  Ports:
  0 [Type:WR, Width:0]:
  0 [Type:RD, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'E.MemFold(a)(0 until 32){i => a }{_+_} {0,0} (NamedTest.scala:40:41)
  1 [Type:WR, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'E.MemFold(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:40:41)
  1 [Type:RD, Width:0]:
  M [Type:WR, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:37:43)
   - Mux Port #1: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:38:43)
   - Mux Port #2: 
    [Ofs: 0]       'E.MemFold(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:39:41)
  M [Type:RD, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:37:43)
   - Mux Port #1: 
    [Ofs: 0]       'B.Reduce(b)(0 until 32){i => a(i) }{_+_} {} (NamedTest.scala:33:38)

Instance #1
  <Banked>
  Depth:    2
  Accum:    Buffer
  Banking:  List(Dims {0}: Cyclic: N=1, B=1, alpha=<1>) <Flat>
  Pipeline: Controller(x62,-1)
  Ports:
  0 [Type:WR, Width:0]:
  0 [Type:RD, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0,0} (NamedTest.scala:37:43)
  1 [Type:WR, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:37:43)
  1 [Type:RD, Width:0]:
  M [Type:WR, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:38:43)
   - Mux Port #1: 
    [Ofs: 0]       'E.MemFold(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:39:41)
   - Mux Port #2: 
    [Ofs: 0]       'E.MemFold(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:40:41)
  M [Type:RD, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'E.MemFold(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:40:41)

Instance #2
  <Banked>
  Depth:    2
  Accum:    Buffer
  Banking:  List(Dims {0}: Cyclic: N=1, B=1, alpha=<1>) <Flat>
  Pipeline: Controller(x88,-1)
  Ports:
  0 [Type:WR, Width:0]:
  0 [Type:RD, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'E.MemFold(a)(0 until 32){i => a }{_+_} {0,0} (NamedTest.scala:39:41)
  1 [Type:WR, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'E.MemFold(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:39:41)
  1 [Type:RD, Width:0]:
  M [Type:WR, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:37:43)
   - Mux Port #1: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:38:43)
  M [Type:RD, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:38:43)

Instance #3
  <Banked>
  Depth:    2
  Accum:    Buffer
  Banking:  List(Dims {0}: Cyclic: N=1, B=1, alpha=<1>) <Flat>
  Pipeline: Controller(x75,-1)
  Ports:
  0 [Type:WR, Width:0]:
  0 [Type:RD, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0,0} (NamedTest.scala:38:43)
  1 [Type:WR, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:38:43)
  1 [Type:RD, Width:0]:
  M [Type:WR, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'D.MemReduce(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:37:43)
   - Mux Port #1: 
    [Ofs: 0]       'E.MemFold(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:39:41)
  M [Type:RD, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'E.MemFold(a)(0 until 32){i => a }{_+_} {0} (NamedTest.scala:39:41)
   - Mux Port #1: 
    [Ofs: 0]       'A.Foreach(0 until 32){i => println(a(i)) } {} (NamedTest.scala:31:44)
   - Mux Port #2: 
    [Ofs: 0]       'B.Reduce(b)(0 until 32){i => a(i) }{_+_} {} (NamedTest.scala:34:38)

Instance #4
  <Banked>
  Depth:    1
  Accum:    None
  Banking:  List(Dims {0}: Cyclic: N=1, B=1, alpha=<1>) <Flat>
  Pipeline: ---
  Ports:
  0 [Type:WR, Width:0]:
  0 [Type:RD, Width:1]:
   - Mux Port #0: 
    [Ofs: 0]       'A.Foreach(0 until 32){i => println(a(i)) } {} (NamedTest.scala:30:44)
   - Mux Port #1: 
    [Ofs: 0]       val sum = 'C.Fold(0)(0 until 32){i => a(i) }{_+_} {} (NamedTest.scala:35:46)
```

dkoeplin, 2018-06-01:
Disregard previous comment about depth needing to be 2 - there's an issue in the MemReduce stage logic still which considers the read of the output of the map and the reduction two separate stages.