stanford-ppl/spatial #33: MemFold into Buffered Memory

mattfel1, 2018-05-30:
I'm trying to figure out what the intention is for a case like in BufferedWrite2:

```
    Accel {

      val wt = SRAM[Int](T)
      val in = SRAM[Int](T)
      Sequential.Foreach(N by T){i =>
        wt load weights(i::i+T)
        in load inputs(i::i+T)

        Foreach(I by 1){x =>
          val niter = Reg[Int]
          niter := x+1
          MemFold(wt)(niter by 1){ k =>  // s0 write
            in
          }{_+_}
          val dummyReg1 = Reg[Int]
          val dummyReg2 = Reg[Int]
          val dummyReg3 = Reg[Int]
          Foreach(T by 1) { i => dummyReg1 := in(i)} // s1 do not touch
          Foreach(T by 1) { i => dummyReg2 := wt(i)} // s2 read
          Foreach(T by 1) { i => dummyReg3 := in(i)} // s3 do not touch
          weightsResult(i*I+x*T::i*I+x*T+T) store wt //s4 read
        }
      }

    }
```

`wt` is triple buffered, and `I` is set to 5. This means that all 3 buffers of `wt` get the original data from the broadcast write which is `wt load weights`.  Then, on iterations:
* `x=0`, `wt = original wt + in`, into buffer 0
* `x=1`, `wt = original wt + 2*in`, into buffer 1
* `x=2`, `wt = original wt + 3*in`, into buffer 2
* `x=3`, `wt = buffer 0 wt + 4*in`

Looking at this code, I don't think the user would be able to know right away that it is a triple buffered memory and things may behave unexpectedly on the fourth iteration, or they may actually just intend for the Foreach to be sequential.  Should this be a warning or what?

dkoeplin, 2018-05-30:
Ah ok, something isn't right here. I believe the correct behavior we settled on here should be to create two duplicates - one local accumulator for the MemFold and one buffer to keep the intermediate values. I believe that should give the correct result. This ties in with the other issue I'm working on with weird instance merging behavior.

dkoeplin, 2018-06-02:
Fixed

commit 610db897 (2018-06-02): Fix #33
closed 2018-06-02 by dkoeplin