stanford-ppl/spatial #35: New Buffer Depth Issue (BufferedWrite2)

mattfel1, 2018-06-02:
I'm looking at BufferedWrite2 and it is now making two duplicates of `wt`, both of depth 1.  It correctly identifies one of the writes as broadcasting though.

dkoeplin, 2018-06-04:
Couldn't reproduce with that app, I'm seeing one instance with depth 1 (the accum) and one with depth 3. Perhaps this was with something I hadn't pushed yet?

mattfel1, 2018-06-04:
Oh whoops.  The buffers are correct.  The actual issue is that there are two writers to it that look like this.  One has none buffer port and the other has a buffer port.  I was assuming that if something is not buffered, then there will be a unique port map for it, ignoring the buffer port.  Will things actually look like this now?

```
                          x328 = SRAMBankedWrite(x303,Vector(x326),Vector(List(Const(0: class emul.FixedPoint))),Vector(b323),ArrayBuffer(Set(b324)))
                           - Type: Void
                           - SrcCtx: BufferedWrite2.scala:32:12
                           - class argon.Consumers: Consumers(Set(x329))
                           - class argon.Effects: (reads={x303}, writes={x303})
                           - class argon.ShallowAliases: ShallowAliases(Set(x328))
                           - class spatial.data.Ports: Ports(Map(List() -> Port(None,0,1,0,0)))
                           - class spatial.data.ScopeCtrl: ScopeCtrl(x329 (scope: 0, 0))
                           - class spatial.data.Dispatch: Dispatch(Map(List() -> Set(0)))
                           - class spatial.data.ReadUses: ReadUses(Set())
                           - class argon.DeepAliases: DeepAliases(Set())
                           - class spatial.data.ParentCtrl: ParentCtrl(x329 (stage: -1))
                           - class spatial.data.ParentBlk: ParentBlk(x329 (block: 0))
                                x380 = SRAMBankedWrite(x303,Vector(x378),Vector(List(Const(0: class emul.FixedPoint))),Vector(b366),ArrayBuffer(Set(b367, b354, b309)))
                                 - Type: Void
                                 - SrcCtx: BufferedWrite2.scala:40:12
                                 - class argon.Consumers: Consumers(Set(x381))
                                 - class argon.Effects: (reads={x303}, writes={x303})
                                 - class argon.ShallowAliases: ShallowAliases(Set(x380))
                                 - class spatial.data.Ports: Ports(Map(List() -> Port(Some(0),0,1,0,0)))
                                 - class spatial.data.ScopeCtrl: ScopeCtrl(x381 (scope: 0, 0))
                                 - class spatial.data.Dispatch: Dispatch(Map(List() -> Set(0)))
                                 - class spatial.data.ReadUses: ReadUses(Set())
                                 - class argon.DeepAliases: DeepAliases(Set())
                                 - class spatial.data.ParentCtrl: ParentCtrl(x381 (stage: -1))
                                 - class spatial.data.ParentBlk: ParentBlk(x381 (block: 0))
```

mattfel1, 2018-06-04:
Another instance of this is SGD, memory sgdmodel_0.  It is not buffered but it has one broadcasting writer and one writer to port 1, neither of these seem to make sense

mattfel1, 2018-06-04:
Probably another angle on the same issue.. The buffered duplicate of wt in BufferedWrite is slightly different.  It has a depth of 3.  Its broadcasting write and its write that should go on buffer port 1, maybe, are both assigned to port 0.  The read is on port 2.

dkoeplin, 2018-06-05:
Looks like the issue here is we only track the port(s) for a given access based on the unrolling factor(s) of the access, not a per-instance set of ports. So writes that go to multiple instances wind up losing info. Should be a pretty easy fix, but it'll change the way port metadata is accessed a bit (need to also have the dispatch ID now)

commit 81292738 (2018-06-05): Fix #35
closed 2018-06-05 by dkoeplin