stanford-ppl/spatial #41: Move SimpleStruct from Outer Control to Inner Control

mattfel1, 2018-06-18:
In Kmeans, there is this snippet:

```
            val minCent = Reduce(accum)(K par PX){ct =>
              val dist = Reg[T](0.to[T])
              Reduce(dist)(D par P2){d => (pts(pt,d) - cts(ct,d)) ** 2 }{_+_}
              pack(ct, dist.value)
            }{(a,b) =>
              mux(a._2 < b._2, a, b)
            }
```

The problem is that the map function for the Reduce is two stages, Reduce and then pack/write to reg.  The SimpleStruct that packs (ct, dist.value) is sitting up in the outer control, which messes up the reg chain logic for bound syms.  If it moves to the inner pipe then cg would properly recognize that it needs to get ct from the second stage of the reg chain

dkoeplin, 2018-06-26:
Changed SimpleStruct to always have to go in InnerPipes (ae3c855f97c915f6313c3ca02f5a63be3e3fc976) - can see if this fixes it

mattfel1, 2018-06-26:
It looks like this properly detects it but now it crashes at treegen because `illegal mix of primitives and control:`.  Testing with Kmeans_slow

dkoeplin, 2018-06-29:
Should be fixed now - added another TransientCleanup pass in the compiler that should take care of this.

closed 2018-06-29 by dkoeplin