stanford-ppl/spatial #46: IfThenElse Returning String

mattfel1, 2018-06-22:
```
val input_file_name = if (!run_on_board) "/home/jcamach2/spatial-lang/apps/src/Rosetta/3D-Rendering/input_triangles.csv" else "/home/jcamach2/Rendering3D/input_triangles.csv"
```
gives the error 
```
[error] /local/ssd/home/mattfel/spatial/test/spatial/tests/feature/unit/hotfix.scala:211:25: ambiguous reference to overloaded definition,
[error] both method __ifThenElse in trait SpatialVirtualization of type [A](cond: spatial.lang.Bit, thenBr: => argon.Sym[A], elseBr: => argon.Sym[A])(implicit ov3: utils.Overloads.Overload3, implicit ctx: forge.SrcCtx, implicit state: argon.State)A
[error] and  method __ifThenElse in trait SpatialVirtualization of type [A](cond: spatial.lang.Bit, thenBr: => argon.Literal, elseBr: => argon.Sym[A])(implicit evidence$4: argon.Type[A], implicit ov2: utils.Overloads.Overload2, implicit ctx: forge.SrcCtx, implicit state: argon.State)A
[error] match argument types (Boolean,String,String)
[error] 		val input_file_name = if (!run_on_board) "/home/jcamach2/spatial-lang/apps/src/Rosetta/3D-Rendering/input_triangles.csv" else "/home/jcamach2/Rendering3D/input_triangles.csv"
[error] 		                      ^
```
This is a reminder for me or someone to look into this

dkoeplin, 2018-06-26:
Confirmed, but not sure why this is happening. Literal should have lower priority than Sym[Text] here

dkoeplin, 2018-06-27:
Similar problem for Boolean / Bit. Looks like any time there's exactly one conversion from unstaged type to Sym, we don't want to make the Literal implicit also available.

commit 4298e74e (2018-06-27): Default runtimeArgs to NoArgs. Fix #46
closed 2018-06-27 by dkoeplin