stanford-ppl/spatial #47: Wrongly Calling Writes Broadcasting

mattfel1, 2018-06-25:
In AES, the writes to `plaintext_sram_0` are mostly broadcasting but none of them should be broadcasting (bufferPort = none)

mattfel1, 2018-06-25:
The issue here was with `findMetaPipe`.  It finds the metapipe LCA, but then when it starts computing dist between all pairs of accesses, it computes this dist relative to the LCA of the two accesses in question rather than going with the metapipe LCA of everyone.  I added a new method called `coarseDistance` that takes the LCA and computes the dist between two accesses relative to this node.  Since the ambiguous metapipes check happened already, we are guaranteed that when we assign the dist of the access, it will either be >= 0 or the access in question will not be contained in the `findAllMetaPipes(readers, writers)` values and we can give it None.

I'm leaving this issue open though because fixing issues #41 and #42 properly will probably change this fix around.

dkoeplin, 2018-06-28:
Reworked this a bit, but your fix looks good. Just unified it a bit with the other methods. 

It looks like `plaintext_sram` is generating 5 instances right now, one for each reader. I think this was the case with the old version of Spatial, but is this expected/desired?

mattfel1, 2018-06-28:
This looks expected.  Old spatial used to have 6 instances but not sure where the extra one was coming from

closed 2018-06-29 by dkoeplin