stanford-ppl/spatial #59: Unused Writes Causing Crash in Transient Cleanup

mattfel1, 2018-07-10:
Although the app designer is the naughty one for making a memory that is never read, we probably shouldn't crash.  SGD and AES are examples

`*** is defined here but never read. Unused writes will be dropped.`

mattfel1, 2018-07-10:
Something with 
```
  override protected def inlineBlock[T](b: Block[T]): Sym[T] = {
    advanceBlk() // Advance block counter before transforming inputs

    withBlockSubsts(b.result) {
      inlineWith(b){stms =>
        stms.foreach(visit)
        withBlockSubsts(b.result){ f(b.result) } // Have to call again in case subst was added inside block
      }
    }
  }
```
in TransientCleanup.  If the `b.result` is actually Invalid (in the case of SGD it is a RegWrite node, which follows another RegWrite to a different reg), it still attempts to create the block with that same result.  I'm not sure what the correct solution is

dkoeplin, 2018-07-16:
Should be fixed now. Using Void constant as the return value of a block if it matches the return type. Fixes bug with no change in program meaning/functionality.

closed 2018-07-16 by dkoeplin