stanford-ppl/spatial #68: Flow Setting Controller as Sequential Before Unit Pipe Insertion

mattfel1, 2018-07-23:
For example:

```
'PROBLEM.Foreach(N by 1){ i =>
  Foreach(...){...}
  sram(i) = ...
}
```
When this gets staged, `PROBLEM` only has one child and therefore its schedule is set to be sequential, even though the sram write will eventually be put in a unit pipe and then it will contain two stages that should be scheduled in a pipeline

dkoeplin, 2018-07-23:
Ah, good call. The analysis should also check if there are any primitive operations directly under that controller that will become other stages.

dkoeplin, 2018-07-23:
Assigned to myself but if you get to it before I do, should be an easy fix

mattfel1, 2018-07-23:
I thought it may be expensive to check every statement in every flow so I
made another case locally where I convert sequenced to pipelined if there
are more than one child, it’s an outer pipe and has no user schedule. This
eventually catches the case in unrolling I think. Maybe it’s a dangerous
assumption but I couldn’t think of any example. Should I just redo it to
use your method?

closed 2018-07-25 by mattfel1