stanford-ppl/spatial #72: negative array crash

shadjis, 2018-07-27:
[crash.txt](https://github.com/stanford-ppl/spatial/files/2236356/crash.txt)

Attached small app crashes with

`[error] (run-main-0) java.lang.NegativeArraySizeException
[error] java.lang.NegativeArraySizeException
[error]         at java.util.Arrays.copyOf(Arrays.java:3332)
[error]         at org.antlr.v4.runtime.ANTLRInputStream.load(ANTLRInputStream.java:101)
[error]         at org.antlr.v4.runtime.ANTLRInputStream.<init>(ANTLRInputStream.java:64)
[error]         at org.antlr.v4.runtime.ANTLRInputStream.<init>(ANTLRInputStream.java:60)
[error]         at org.antlr.v4.runtime.ANTLRInputStream.<init>(ANTLRInputStream.java:68)
[error]         at firrtl.Parser$$anonfun$1.apply(Parser.scala:35)
[error]         at firrtl.Parser$$anonfun$1.apply(Parser.scala:29)
`

mattfel1, 2018-08-16:
Not getting this crash in broadcast branch.  I'm not confident the issue is fixed, but I think it just happens to generate something small enough to not have this issue.

shadjis, 2018-08-21:
Crash is gone for this app, but still exists on other apps
Closing as duplicate of #75

closed 2018-08-21 by shadjis