stanford-ppl/spatial #73: par not speeding up

shadjis, 2018-07-27:
[dp_simple.txt](https://github.com/stanford-ppl/spatial/files/2236424/dp_simple.txt)

attached app is simple dot product

unparallelized the dot product loop (ln 33) takes around 1024 cycles
with par 4 loop takes 538
I think it should be more like 256
is this difference expected

shadjis, 2018-07-30:
maybe I am doing something wrong because I pulled on issue61_cycle_detection and this went from 538 to 6403 cycles now?

mattfel1, 2018-07-30:
Right. This is what I noticed too.  I just unbroke the optimizedReduce flag
on issue61 branch.  It turns out it was interacting badly with the IterDiff
stuff.  The 538 cycles vs 256 cycles is due to the FixAdd cycle.  This has
an II of 2 and I didn't put in the hack for FixAdd reductions (I only did
FixFMA reductions), so this would account for the slowdown of 2x.  Will see
if I can get accum branch wrapped up by tomorrow

shadjis, 2018-07-30:
I pulled and I still get 6403, should it be 538 again now?

mattfel1, 2018-07-30:
Oops, pull again

mattfel1, 2018-08-07:
The loop takes 286 cycles on accum branch now

mattfel1, 2018-08-16:
works on master and broadcast branches now

closed 2018-08-16 by mattfel1