stanford-ppl/spatial #75: problem app

shadjis, 2018-07-30:
[problem_app.txt](https://github.com/stanford-ppl/spatial/files/2243074/problem_app.txt)

Attached is a simple example app which reveals some problems:

1. If outP = 1 and P2 = 32, it crashes in `make vcs` with  `java.lang.OutOfMemoryError`
2. If outP = 2 and P2 = 16, it crashes in `make vcs` with the same error as #72 
3. If outP = 4 and P2 = 8, it passes but takes long to `make vcs`. Maybe this long time is expected with these pars but they are not that high so I just wanted to check because the app is not that complicated but it takes long.

All 3 have the same total par, and I would expect errors to occur at higher outP, not lower

mattfel1, 2018-07-30:
I've been staring at this for a while now and I think at least problem #1 could be related to an extremely large number of wires getting generated, for example in patterns like 
```
DataAsBits
VecSlice
VecApply
DataAsBits
VecApply
DataAsBits
....
VecApply
VecConcat
BitsAsData
```

Another source of abundant wires could be from the way vectorized sram reads are split into individual lanes in cg.  I'll see what I can do about this one now and then look at if there is a more concise way to capture the first thing

mattfel1, 2018-08-01:
awaiting #80

mattfel1, 2018-08-03:
#80 was merged but there still seems to be some java limitations so leaving this issue open for now until we find either more opportunities for compressing the code or (hopefully) the specialized reduction + broadcasting reads shrinks the code siginificantly

shadjis, 2018-08-17:
I'm not sure if this is useful but this gives a different error now

`[error] (run-main-0) java.lang.IndexOutOfBoundsException: 32
[error] java.lang.IndexOutOfBoundsException: 32
[error]         at scala.collection.immutable.Vector.checkRangeConvert(Vector.scala:132)
[error]         at scala.collection.immutable.Vector.apply(Vector.scala:122)
[error]         at chisel3.core.Vec.apply(Aggregate.scala:221)
`

shadjis, 2018-08-21:
The error above is gone after 307fa5b but the original errors are back now

mattfel1, 2018-10-08:
Refactoring chisel gen to hopefully solve this permanently (there still may be a mem limit issue but should be strictly forward progress anyway).  See #134

closed 2018-10-08 by mattfel1