stanford-ppl/spatial #76: Update Documentation / Getting Started Tutorial

ZhijieWang, 2018-07-31:
Hi,

Trying to run the getting started tutorial and experience some difficulty. Seems the tutorial is slightly outdated. Can someone point me to the correct one? 

Thank you
Bill Wang

```
➜  spatial-lang git:(master) ✗ bin/spatial HelloSpatial  
sbt -batch "apps/run-main HelloSpatial "
[info] Loading global plugins from /Users/zhijie.wang/.sbt/0.13/plugins
[info] Loading project definition from /Users/zhijie.wang/Developer/spatial-lang/project
[info] Set current project to spatial-lang (in build file:/Users/zhijie.wang/Developer/spatial-lang/)
[info] Compiling 7 Scala sources to /Users/zhijie.wang/Developer/spatial-lang/apps/target/scala-2.12/classes...
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:13: Type Int is not a staged type. Try adding an explicit lift() call?
[error]     val argin1 = ArgIn[Int]   // Register that is written to by the host and read from by the Accel
[error]                       ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:14: Type Int is not a staged type. Try adding an explicit lift() call?
[error]     val argout1 = ArgOut[Int] // Register that is written to by the Accel and read from by the host
[error]                         ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:15: Type Int is not a staged type. Try adding an explicit lift() call?
[error]     val io1 = HostIO[Int]     // Register that can be both written to and read from by the Accel and the host
[error]                     ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:20: not found: value SourceContext
[error]     val argin2 = ArgIn[T]
[error]                       ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:22: not found: value SourceContext
[error]     setArg(argin1, args(0).to[Int]) // Set argument with the first command-line value
[error]                         ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:23: not found: value SourceContext
[error]     setArg(argin2, 7.to[T]) // Args do not necessarily need to be set with command-line values
[error]                        ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:23: not found: value SourceContext
[error]     setArg(argin2, 7.to[T]) // Args do not necessarily need to be set with command-line values
[error]           ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:24: not found: value SourceContext
[error]     setArg(io1, args(1).to[Int])
[error]                      ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:28: not found: value SourceContext
[error]     val data2D        = (0::64, 0::64){(i,j) => i*100 + j} // Create 64x64 2D, where each element is row * 100 + col
[error]                           ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:29: not found: value SourceContext
[error]     val data5D        = (0::2, 0::2, 0::2, 0::2, 0::16){(i,j,k,l,m) => random[Int](5)} // Create 5D tensor, the highest dimension tensor currently supported in Spatial, with each element a random Int between 0 and 5
[error]                           ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:29: could not find implicit value for evidence parameter of type argon.lang.direct.Bits[Int]
[error]     val data5D        = (0::2, 0::2, 0::2, 0::2, 0::16){(i,j,k,l,m) => random[Int](5)} // Create 5D tensor, the highest dimension tensor currently supported in Spatial, with each element a random Int between 0 and 5
[error]                                                                                   ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:31: not found: value SourceContext
[error]     val dram1D        = DRAM[Int](64)
[error]                                   ^
[info] Int(64) <: spatial.lang.Index?
[info] false
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:31: Type Int is not a staged type. Try adding an explicit lift() call?
[error]     val dram1D        = DRAM[Int](64)
[error]                                  ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:32: not found: value SourceContext
[error]     val dram1D_longer = DRAM[Int](1024)
[error]                                   ^
[info] Int(1024) <: spatial.lang.Index?
[info] false
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:32: Type Int is not a staged type. Try adding an explicit lift() call?
[error]     val dram1D_longer = DRAM[Int](1024)
[error]                                  ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:33: not found: value SourceContext
[error]     val dram2D        = DRAM[Int](64,64)
[error]                                   ^
[info] Int(64) <: spatial.lang.Index?
[info] false
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:33: not found: value SourceContext
[error]     val dram2D        = DRAM[Int](64,64)
[... 96 lines cut ...]
[error]     println("2D pass? " + cksum_2D)
[error]            ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:77: not found: value SourceContext
[error]     println("scatter pass? " + cksum_scattered)
[error]            ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:83: not found: value SourceContext
[error]     println("Received " + result1 + " and " + result2)
[error]            ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:84: not found: value SourceContext
[error]     val cksum = (result1 == {args(0).to[Int] + args(1).to[Int]}) && (result2 == args(0).to[Int]) // The {} brackets are Scala's way of scoping operations
[error]                                   ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:84: not found: value SourceContext
[error]     val cksum = (result1 == {args(0).to[Int] + args(1).to[Int]}) && (result2 == args(0).to[Int]) // The {} brackets are Scala's way of scoping operations
[error]                                                     ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:84: not found: value SourceContext
[error]     val cksum = (result1 == {args(0).to[Int] + args(1).to[Int]}) && (result2 == args(0).to[Int]) // The {} brackets are Scala's way of scoping operations
[error]                                                                                      ^
[error] /Users/zhijie.wang/Developer/spatial-lang/apps/src/HelloSpatial.scala:85: not found: value SourceContext
[error]     println("ArgTest pass? " + cksum)
[error]            ^
[error] 44 errors found
[error] (apps/compile:compileIncremental) Compilation failed
[error] Total time: 75 s, completed Jul 31, 2018 3:29:58 PM
[error] Spatial compilation failed
```

dkoeplin, 2018-07-31:
Hi Bill, I just copied and pasted the code off the tutorial/documentation site and compiled it with no issues. Perhaps there's something that got copied incorrectly with the source file you're using? The Scala compiler is giving errors about `Int` being unstaged, which suggests part of Spatial isn't being imported properly to shadow `scala.Int`, and that SourceContext is missing, which is created by an implicit method imported in Spatial. Maybe the import statement `import spatial.dsl._` is missing/incorrect?

ZhijieWang, 2018-08-01:
after wiping spatial-quickstart and spatial-lang then reclone everything, the tutorial works. Thanks.

closed 2018-08-01 by ZhijieWang