stanford-ppl/spatial #84: sim and synth mismatch

shadjis, 2018-08-05:
Simple app sim_synth_mismatch_issue84.scala has a mismatch in sim and synth. May be related to #82 or #83 because app is similar, but unlike #82 which mismatches with parallelism, this app does not contain parallelism. The first few outputs should be 0.096191406

mattfel1, 2018-08-21:
0.096191	0.096191	0.096191	0.096191	0.096191	0.096191	0.096191	0.096191	0.096191	0.096191	0.096191	0.096191	
-9.439453	-14.985352	-1.245605	7.139160	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	
0.096191	0.089844	0.337402	0.545898	0.493164	0.504395	0.481934	0.489746	0.291504	0.096191	0.096191	0.096191	
1.786133	-3.911133	-12.722656	-11.981445	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	
0.096191	0.473145	0.479980	0.249512	0.288086	0.285156	0.534180	0.474121	0.233398	0.096191	0.096191	0.096191	

I'm getting these results now.  Do these look good enough?

shadjis, 2018-08-21:
yes it matches now

closed 2018-08-21 by shadjis