stanford-ppl/spatial #90: Port Info Conflicts On RegRead When Unrolling Inside Of Switch

mattfel1, 2018-08-10:
Here is a test app:
```
@spatial class SwitchCondReuse extends SpatialTest {

  def main(args: Array[String]): Unit = {
    val x = ArgIn[Int]
    val in = args(0).to[Int]
    setArg(x,in)
    val out = DRAM[Int](16)

    Accel {
      val sram = SRAM[Int](16)
      if (x.value < 5) {
        Foreach(16 by 1 par 4){i => 
          Pipe{sram(i) = i}
          Pipe{sram(i) = i}
        }
      }
      out store sram
    }

    val gold = Array.tabulate(16){i => i}
    val got = getMem(out)
    assert(gold == got)
    printArray(gold, "Gold")
    printArray(got, "Got")
    println(r"PASS: ${gold == got}")
  }

}
```

There are multiple RegReads to x102 on Port(Some(0),0,1,0,0)

mattfel1, 2018-08-10:
Has to do with something like this:

Before Unrolling:
```
                          x116 = RegRead(x102)
                           - Aliases: 0028: x107
                           - class spatial.metadata.memory.Ports: Ports(Map(9 -> Map(List() -> Port(Some(0),0,1,0,0))))
```

During Unrolling:
```
            Access: x116
            Memory: x102
            Iterators between x116 and x102: 
            Unrolling x116 = RegRead(x102)
             - Aliases: 0028: x107
             - Type: Bit
             - SrcCtx: UnrollingTest.scala:40:19
             - class spatial.metadata.memory.Dispatch: Dispatch(Map(List() -> Set(9)))
             - class argon.ShallowAliases: ShallowAliases(Set(x116))
             - class spatial.metadata.access.AffineMatrices: AffineMatrices(1 matrices)
             - class spatial.metadata.control.ParentCtrl: ParentCtrl(x13 (stage: -1))
             - class spatial.metadata.access.AccessPattern: AccessPattern(List())
             - class argon.Consumers: Consumers(Set(x13, x16))
             - class spatial.metadata.access.ReadUses: ReadUses(Set(x116))
             - class argon.DeepAliases: DeepAliases(Set())
             - class spatial.metadata.memory.Ports: Ports(Map(9 -> Map(List() -> Port(Some(0),0,1,0,0))))
             - class spatial.metadata.control.ScopeCtrl: ScopeCtrl(x13 (scope: -1, -1))
             - class argon.Effects: (unique=true, reads={x102})
             - class spatial.metadata.control.DefiningBlk: DefiningBlk(x16 (block: 0))
              Dispatch: ArrayBuffer(9)
              Lane IDs: ArrayBuffer(0)
              Port:     Port(Some(0),0,1,0,0)
              Masters: ArrayBuffer(0) // Non-duplicated lane indices
              Readers of x139 is now: Set(x171)
              x171 = RegRead(x139)
             - Type: Bit
             - class spatial.metadata.memory.Dispatch: Dispatch(Map(List() -> Set(0)))
             - class spatial.metadata.memory.Ports: Ports(Map(0 -> Map(List() -> Port(Some(0),0,1,0,0))))
              Dispatch: ArrayBuffer(9)
              Lane IDs: ArrayBuffer(1)
              Port:     Port(Some(0),0,1,0,0)
              Masters: ArrayBuffer(1) // Non-duplicated lane indices
              Readers of x139 is now: Set(x171, x172)
              x172 = RegRead(x139)
             - Type: Bit
             - class spatial.metadata.memory.Dispatch: Dispatch(Map(List() -> Set(0)))
             - class spatial.metadata.memory.Ports: Ports(Map(0 -> Map(List() -> Port(Some(0),0,1,0,0))))
              Dispatch: ArrayBuffer(9)
              Lane IDs: ArrayBuffer(2)
              Port:     Port(Some(0),0,1,0,0)
              Masters: ArrayBuffer(2) // Non-duplicated lane indices
              Readers of x139 is now: Set(x171, x172, x173)
              x173 = RegRead(x139)
             - Type: Bit
             - class spatial.metadata.memory.Dispatch: Dispatch(Map(List() -> Set(0)))
             - class spatial.metadata.memory.Ports: Ports(Map(0 -> Map(List() -> Port(Some(0),0,1,0,0))))
              Dispatch: ArrayBuffer(9)
              Lane IDs: ArrayBuffer(3)
              Port:     Port(Some(0),0,1,0,0)
              Masters: ArrayBuffer(3) // Non-duplicated lane indices
              Readers of x139 is now: Set(x171, x172, x173, x174)
              x174 = RegRead(x139)
             - Type: Bit
             - class spatial.metadata.memory.Dispatch: Dispatch(Map(List() -> Set(0)))
             - class spatial.metadata.memory.Ports: Ports(Map(0 -> Map(List() -> Port(Some(0),0,1,0,0))))
```

After Unrolling, all 4 of these unrolled regreads got dispatched to the same place and they all kept the same port info.  I'm pretty sure this will be fixed by #81 but for now I'm just going to do this hack in the unroller that checks for port collisions during unrolling and bumps the port of the unrolled instance to the next available one

dkoeplin, 2018-08-15:
Should also be closed with the fix to #94

closed 2018-08-15 by dkoeplin