stanford-ppl/spatial #91: Synth Out of Memory

ZhijieWang, 2018-08-13:
Hi,

What is the recommended machine (RAM) setting and SBT config?
Currently encountering this OOM issue on a server of 2X CPU (8 core) and 128GB Memory.
```
diuser@infra-1:~/spatial$  bin/spatial InOutArg --synth 

[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/diuser/spatial/project
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Set current project to nova (in build file:/home/diuser/spatial/)
[info] Set current project to apps (in build file:/home/diuser/spatial/)
[success] Total time: 2 s, completed Aug 13, 2018 2:05:48 PM
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/diuser/spatial/project
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Set current project to nova (in build file:/home/diuser/spatial/)
[info] Compiling 218 Scala sources to /home/diuser/spatial/target/scala-2.12/test-classes ...
java.lang.OutOfMemoryError: GC overhead limit exceeded
	at scala.tools.nsc.backend.jvm.BTypesFromSymbols.methodBTypeFromMethodType(BTypesFromSymbols.scala:128)
	at scala.tools.nsc.backend.jvm.BTypesFromSymbols.methodBTypeFromSymbol(BTypesFromSymbols.scala:118)
	at scala.tools.nsc.backend.jvm.BCodeSkelBuilder$PlainSkelBuilder.genDefDef(BCodeSkelBuilder.scala:553)
	at scala.tools.nsc.backend.jvm.BCodeSkelBuilder$PlainSkelBuilder.gen(BCodeSkelBuilder.scala:508)
	at scala.tools.nsc.backend.jvm.BCodeSkelBuilder$PlainSkelBuilder.$anonfun$gen$7(BCodeSkelBuilder.scala:510)
	at scala.tools.nsc.backend.jvm.BCodeSkelBuilder$PlainSkelBuilder.gen(BCodeSkelBuilder.scala:510)
	at scala.tools.nsc.backend.jvm.BCodeSkelBuilder$PlainSkelBuilder.genPlainClass(BCodeSkelBuilder.scala:110)
	at scala.tools.nsc.backend.jvm.CodeGen.genClass(CodeGen.scala:69)
	at scala.tools.nsc.backend.jvm.CodeGen.genClassDef$1(CodeGen.scala:31)
	at scala.tools.nsc.backend.jvm.CodeGen.$anonfun$genUnit$3(CodeGen.scala:54)
	at scala.tools.nsc.backend.jvm.CodeGen.genClassDefs$1(CodeGen.scala:54)
	at scala.tools.nsc.backend.jvm.CodeGen.$anonfun$genUnit$2(CodeGen.scala:53)
	at scala.tools.nsc.backend.jvm.CodeGen.genClassDefs$1(CodeGen.scala:53)
	at scala.tools.nsc.backend.jvm.CodeGen.$anonfun$genUnit$4(CodeGen.scala:58)
	at scala.tools.nsc.backend.jvm.CodeGen.genUnit(CodeGen.scala:58)
	at scala.tools.nsc.backend.jvm.GenBCode$BCodePhase.apply(GenBCode.scala:67)
	at scala.tools.nsc.Global$GlobalPhase.$anonfun$applyPhase$1(Global.scala:426)
	at scala.tools.nsc.Global$GlobalPhase.applyPhase(Global.scala:419)
	at scala.tools.nsc.Global$GlobalPhase.$anonfun$run$1(Global.scala:390)
	at scala.tools.nsc.Global$GlobalPhase.$anonfun$run$1$adapted(Global.scala:390)
	at scala.tools.nsc.Global$GlobalPhase$$Lambda$3491/1789832272.apply(Unknown Source)
	at scala.collection.Iterator.foreach(Iterator.scala:944)
	at scala.collection.Iterator.foreach$(Iterator.scala:944)
	at scala.collection.AbstractIterator.foreach(Iterator.scala:1432)
	at scala.tools.nsc.Global$GlobalPhase.run(Global.scala:390)
	at scala.tools.nsc.backend.jvm.GenBCode$BCodePhase.super$run(GenBCode.scala:73)
	at scala.tools.nsc.backend.jvm.GenBCode$BCodePhase.$anonfun$run$1(GenBCode.scala:73)
	at scala.tools.nsc.backend.jvm.GenBCode$BCodePhase.run(GenBCode.scala:71)
	at scala.tools.nsc.Global$Run.compileUnitsInternal(Global.scala:1446)
	at scala.tools.nsc.Global$Run.compileUnits(Global.scala:1430)
	at scala.tools.nsc.Global$Run.compileSources(Global.scala:1423)
	at scala.tools.nsc.Global$Run.compile(Global.scala:1539)
[error] Error while emitting ParLaneMask.scala
[error] GC overhead limit exceeded
java.lang.OutOfMemoryError: GC overhead limit exceeded
	at scala.tools.nsc.backend.jvm.BCodeSkelBuilder$PlainSkelBuilder.initJMethod(BCodeSkelBuilder.scala:541)
	at scala.tools.nsc.backend.jvm.BCodeSkelBuilder$PlainSkelBuilder.genDefDef(BCodeSkelBuilder.scala:580)
	at scala.tools.nsc.backend.jvm.BCodeSkelBuilder$PlainSkelBuilder.gen(BCodeSkelBuilder.scala:508)
	at scala.tools.nsc.backend.jvm.BCodeSkelBuilder$PlainSkelBuilder.$anonfun$gen$7(BCodeSkelBuilder.scala:510)
	at scala.tools.nsc.backend.jvm.BCodeSkelBuilder$PlainSkelBuilder.gen(BCodeSkelBuilder.scala:510)
	at scala.tools.nsc.backend.jvm.BCodeSkelBuilder$PlainSkelBuilder.genPlainClass(BCodeSkelBuilder.scala:110)
	at scala.tools.nsc.backend.jvm.CodeGen.genClass(CodeGen.scala:69)
	at scala.tools.nsc.backend.jvm.CodeGen.genClassDef$1(CodeGen.scala:31)
	at scala.tools.nsc.backend.jvm.CodeGen.$anonfun$genUnit$3(CodeGen.scala:54)
	at scala.tools.nsc.backend.jvm.CodeGen.genClassDefs$1(CodeGen.scala:54)
	at scala.tools.nsc.backend.jvm.CodeGen.$anonfun$genUnit$2(CodeGen.scala:53)
	at scala.tools.nsc.backend.jvm.CodeGen.genClassDefs$1(CodeGen.scala:53)
[... 173 lines cut ...]
Model name:            Intel(R) Xeon(R) CPU E5-2620 v4 @ 2.10GHz
Stepping:              1
CPU MHz:               1200.773
CPU max MHz:           3000.0000
CPU min MHz:           1200.0000
BogoMIPS:              4201.55
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              20480K
NUMA node0 CPU(s):     0-7,16-23
NUMA node1 CPU(s):     8-15,24-31
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch epb invpcid_single intel_pt ibrs ibpb stibp kaiser tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm cqm rdseed adx smap xsaveopt cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local dtherm ida arat pln pts
diuser@infra-1:~/spatial$ free -h
              total        used        free      shared  buff/cache   available
Mem:           125G        1.2G        120G         39M        4.0G        123G
Swap:          132G          0B        132G
diuser@infra-1:~/spatial$ uname -a
Linux infra-1 4.4.0-131-generic #157-Ubuntu SMP Thu Jul 12 15:51:36 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
diuser@infra-1:~/spatial$ free -h
              total        used        free      shared  buff/cache   available
Mem:           125G        1.6G        119G         95M        4.1G        123G
Swap:          132G          0B        132G
```

dkoeplin, 2018-08-14:
Looks like you're actually running out of memory while Scala is compiling our regression tests, before Spatial ran. I usually use somewhere between 4 - 8 GB (depending on machine availability) to compile the code. Spatial itself generally doesn't use more than 1 GB, but I usually keep the same settings for running as compiling.

ZhijieWang, 2018-08-16:
For others who: 
```
export JAVA_OPTS="-Xms1024m -Xmx8G"
```
This was able to get me over the compilation step.

@dkoeplin  thanks for the clarification

closed 2018-08-16 by ZhijieWang