stanford-ppl/spatial #100: Overly Aggressive Broadcasting (?)

mattfel1, 2018-08-23:
Took a while, but I think I finally found the issue with MD_Grid.  It works with broadcasting turned off, but fails with it turned on.  We have something like this:

```
MemReduce(b0_cube_forces)(... b1z_start until b1z_end by 1 par 2) { (b1x, b1y, b1z) => 
    val p_range = npoints_sram(b0x, b0y, b0z)
    Foreach(0 until p_range) { p_idx =>
        ...
        Reduce(...)(...) { q_idx =>  // This loop runs for an amount of time that varies with b1z
           ...
        }
    }
}
```

I'm pretty sure the issue is that it broadcasts the `p_range` read, but this read can happen at different times in each of the unrolled bodies, since the innermost thing can run for a variable amount of time.

dkoeplin, 2018-08-23:
Interesting, the lockstep check should catch this case. It's supposed to
check if any loop bodies exist which may run for a variable amount of time,
but sounds like it's not correctly checking this case

mattfel1, 2018-08-23:
Check the diff for 2b0c446613abbed264b1bbcb601a93ffdab97bea

Changes are:
1) When checking lockstep, include all children if isLoop, otherwise only check prior to the appropriate childe
2) Use the Sym rather than the scope for checking isLockstepAcross, since the scope wasn't finding children who exist

I think this is working for MD_Grid, the question is if any other opportunities for broadcasting are missed now.  The Broadcast app seems like it still broadcasts correctly but not sure if this makes other apps miss the chance.

closed 2018-08-31 by mattfel1