stanford-ppl/spatial #118: One gatherBuffer in Fringe can cause another one to hang

mattfel1, 2018-09-01:
In SPMV_DumbPack, there is a gather inside of a loop with par 2.  Because of dram transfers that happen prior to the gather, the two gathers start at different times.  There is a little bit of address padding at the end to align the gathers (60 elements requested gets padded with + 4 extra requests from mem(0)).  The problem is that the first gather issues these requests and gets the response.  In the middle of the responses, the second gather requests the same padding addresses just in time to see some of the responses but not all of them, so it hangs

mattfel1, 2018-09-01:
[image]
The first pair of signals shows the addr for bank 60 in the first gather buffer, along with the boolean tracking when it collects incoming data.  The second pair shows addr for bank 60 in the second gather buffer, which gets set and misses some of the incoming data

matthewvilim, 2018-09-07:
fixed at https://github.com/stanford-ppl/spatial/commit/7a0755bb4e407daca6dfc8aa3cf7b8d09da8b34b

closed 2018-09-07 by matthewvilim