stanford-ppl/spatial #138: Host vs. Accel DRAM syntax is confusing

dkoeplin, 2018-10-12:
Noticed the new distinction between accel and host DRAM. While this is good, the way of creating these is not at all clear. At the moment, calling ```DRAM[Int](32)``` creates a host DRAM while calling ```DRAM1[Int](32)``` creates an Accel DRAM. 

It would be much better if AccelDRAM was the default (since that's what it previously was), and there was an option to do something like ```HostDRAM[Int](32)``` or ```DRAM.host[Int](32)```. Something that makes it obvious this is a different DRAM type.

matthewvilim, 2018-10-12:
Just to clarify, the current API is:

// dram of statically known dimensions that is allocated by the host and passed in with implicit ArgIns to be shared with the accelerator; this hasn't changed at all from the previous design. It's staged as a DRAMHost node
val host = DRAM[Int](dims...)

// dram of fixed rank but variable dimensions; it's inaccessible to the host and is managed by the accelerator; the current implementation of the hardware heap pokes some registers on the host to request an allocation, but eventually it will be managed fully in hardware. It's staged as a DRAMAccel node.
val accel = DRAM1,2,3,4,5[Int]
accel.alloc(dims...)
accel.dealloc

Would you prefer changed the lang api to DRAMAccel1,2,3,4,5 to make it more clear that this DRAM is managed completely by the accelerator?

dkoeplin, 2018-10-12:
That clears things up actually. I think in that case yeah, it would be good just to note that it's accelerator managed memory. Something like ```AccelDRAM``` or ```DynamicDRAM``` would be a nice mnemonic, otherwise it's hard to distinguish from the host-managed DRAM.