stanford-ppl/spatial #151: Auto Schedule Controllers in Parallel if no dependencies

mattfel1, 2018-10-29:
For example:
```
val x = Reduce(Reg[T])(...){i => mem1(i)}{_+_}
val y = Reduce(Reg[T])(...){i => mem2(i)}{_+_}
```
should be scheduled inside a parallel......

Or its time to move to token based control?

dkoeplin, 2018-10-29:
+1 for moving to token-based control. In that case we would need some improved helper methods to track things that will actually execute in parallel and producer/consumer distances.

mattfel1, 2019-05-04:
Done.  There is a --noBindParallels global compile flag, and also a Pipe.NoBind annotation you can add to specific controllers.  Some apps get a tiny speedup, some get a slowdown in vcs depending on how this changes dram congestion but I added the flag to these so they should perform the same as before.

closed 2019-05-04 by mattfel1