stanford-ppl/spatial #162: Mysterious ISL Issue

mattfel1, 2018-11-25:
Seen it in a few places now but have no idea how to reproduce it:

```
[info] Pass: MemoryAnalyzer
[bug] An exception was encountered while compiling Test:
[bug]   Stream closed
[bug]   java.io.IOException: Stream closed
[bug]   java.lang.ProcessBuilder$NullOutputStream.write(ProcessBuilder.java:433)
[bug]   java.io.OutputStream.write(OutputStream.java:116)
[bug]   java.iw.BufferedOutputStream.flushBuffer(BufferedOutputStream.java:82)
[bug]   java.io.BufferedOutputStream.flush(BufferedOutputStream.java:140)
[bug]   sun.nio.cs.StreamEncoder.implFlush(StreamEncoder.java:297)
[bug]   sun.niw.cs.StreamEncoder.flush(StreamEncoder.java:141)
[bug]   java.io.OutputStreamWriter.flush(OutputStreamWriter.java:229)
[bug]   java.io.BufferedWriter.flush(BufferedWriter.java:254)
[bug]   utils.process.BackgroundProcess.send(BackgroundProcess.scala:16)
[bug]   poly.ISL.isEmpty(ISL.scala:115)
[bug]   poly.ISL.isEmpty$(ISL.scala:112)
[bug]   spatial.Spatial$SpatialISL.isEmpty(Spatial.scala:36)
[bug]   poly.ISL.nonEmpty(ISL.scala:110)
[bug]   poly.CSL.nonEmpty$(ISL.scala:110)
[bug]   spatial.Spatial$SpatialISL.nonEmpty(Spatial.scala:36)
[bug]   poly.ISL.overlapsAddress(ISL.scala:130)
[bug]   poly.TSL.overlapsAddress$(ISL.scala:128)
[bug]   spatial.Spatial$SpatialISL.overlapsAddress(Spatial.scala:36)
 bug]   spatial.metadata.access.AccessMatrix.overlapsAddress(AffineData.scala:38)
 bug]   spatial.traversal.banking.MemoryConfigurer.accessesConflict(MemoryConfigurer.scala:515)
[bug]  .. [see /armadillo/users/ctimothy/fpga-grazelle/spatial-quickstart/logs/Test/Test_exception.log]
 bug] This is due to a compiler bug. A log file has been created at:
 bug]   /armadillo/users/ctimothy/fpga-grazelle/spatial-quickstart/logs/Test/Test_exception.log
[warn] 1 warning found
[failed] Total time: 1.8860 seconds
```

dkoeplin, 2018-11-27:
Just ran into this as well. I'm not sure why the external process is shutting down, but gonna try a workaround to just reopen it when this happens

mattfel1, 2018-11-27:
Yay!  Students have seen this on corn and on their own computers but I've never been able to reproduce it anywhere, even when I installed on corn myself.

closed 2019-02-11 by mattfel1