stanford-ppl/spatial #166: Fork Inside ForkJoin With Pipe Insertion (Reprise)

mattfel1, 2018-11-30:
There should really be a fix for this kind of thing since its not obvious to the average user and it always takes a few minutes to trace down the issue for me when I'm debugging their waveform.
```
Parallel{
  val cond = // sram read
  if (cond) { // Outer switch
    ...
  }
  if (cond) { // Outer switch // Second one so it matches with the image below
    ...
  }
}
```
becomes
[image]

mattfel1, 2018-12-04:
Fixed this in what I thought was the most reasonable way.  Generally people tend to write if blocks inside of their Parallel and they actually intend for the cond resolution to happen before the switch runs.  (also happens with setting up dram transfer bounds).  The pipe inserter used to step through the stms of an outer controller and mark the stage that each one belongs in.  Then it would go through and transform each stage, inserting the escaping regs if needed.  I added a step inbetween here where it steps through each of the stages and binds one stage to its subsequent stage if the escaping values are used there, and wraps the two together in a pipe.

What I'm not sure about is if there is a condition used by multiple stages in the parallel.  Will the transient analysis have already given each stage its own read of the condition, and will they be placed in the IR in the order that makes this new check work?

closed 2018-12-04 by mattfel1

dkoeplin, 2018-12-04:
Your fix makes sense, but technically the original program is incorrect. Parallel means do everything in parallel, so including things that had data dependencies is an incorrect usage. 

The correct implementation should be (and perhaps we can automate it to this): 
```
// Should get a Pipe { } and a register
val cond = // sram read

Parallel{  
  if (cond) { // Outer switch
    ...
  }
  if (cond) { // Outer switch // Second one so it matches with the image below
    ...
  }
}
```