stanford-ppl/spatial #168: Warning for Const out of range Missing

mattfel1, 2018-12-03:
We had this case:

`local_edges_l(container_begin + edge_iter) & 0x8000000000000000l`

where local_edges_l is `FixPt[TRUE,_64,_0]`.  It didn't warn about 0x8000000000000000 being out of range, but the generated verilog had 0x7fffffffffffff secretly, which defeats the whole purpose of this mask.  Why were we not warned?

dkoeplin, 2018-12-03:
ah... the detection for the warning is percentage based right now... meaning big numbers won't get the warning for small differences like this, even though they're being used as bit masks. Clearly that's not the correct solution to find these cases..

mattfel1, 2018-12-04:
Got it.  More concerning is that --synth and --sim don't match up in this case.  One or both are doing something suspicious, but either way I will go in and put a warning.