stanford-ppl/spatial #176: Synthesis Issues, AFGEMM

kelayamatoz, 2019-01-07:
AFGEMM app: [app.zip](https://github.com/stanford-ppl/spatial/files/2733687/app.zip)
There are a few issues for this app when synthesizing it using the Xilinx toolchain: 
1. When the size of SRAM gets large, the generated verilog SRAMs are not implemented using block rams, but instead using distributed rams.
2. The reference, hand-written verilog design implements a 16x16 matrix vector multiplication (MVM) in the inner loop. The MVM is iterated on the m dimension to make the full matrix multiplication. To achieve this goal, the inner-most loop is parallelized by 16x16 = 256. The generated verilog file is quite large and xilinx's synthesizer ends up running out of memory on a server with ~300 GB.
3. Xilinx's synthesizer could not utilize parallel synth and impl for the generated verilog of this app.

kelayamatoz, 2019-01-07:
Guesses on each point: 
1. Vivado could not infer block ram from looking at the verilog syntax for memory. We may want to replace the verilog memory implementation with tcl memory IP. 
2. This issue is possibly related to how we generate verilog for the inner-most pipe. In this app, the inner-most pipe is par by 256. A lot of muxes and logics are generated within one verilog module.
3. This issue is related to 2. My guess is that Vivado assigns one module per job, and the job synthesizing the inner-most loop takes too long to finish due to the redundant muxes and logics. We could try to split the large modules into smaller ones and check if this helps Vivado do parallel synth.

In short, I think the root cause of these issues is that we never stretched the support for PCIe FPGA cards. On SoCs, we have limited resources and we could not parallelize a design by a huge factor. However on PCIe cards, we usually have much more resources and we could have much larger parallelization factors.

kelayamatoz, 2019-01-24:
I ran the issue177 branch with the AFGEMM app using --synth --retime --fpga ZCU flags. In gen when I called make, I got the following errors:

[error] /home/tianzhao/spatial/gen/AFGEMM/chisel/sm_x43580.scala:21:20: Platform restriction: a parameter list's length cannot exceed 254.
[error] class x43580_kernel(
[error]                    ^
[error] /home/tianzhao/spatial/gen/AFGEMM/chisel/sm_x43710_outr_Foreach.scala:21:33: Platform restriction: a parameter list's length cannot exceed 254.
[error] class x43710_outr_Foreach_kernel(
[error]                                 ^
[error] /home/tianzhao/spatial/gen/AFGEMM/chisel/sm_x43711_outr_Foreach.scala:21:33: Platform restriction: a parameter list's length cannot exceed 254.
[error] class x43711_outr_Foreach_kernel(
[error]                                 ^
[error] /home/tianzhao/spatial/gen/AFGEMM/chisel/sm_x43716_outr_UnitPipe.scala:21:34: Platform restriction: a parameter list's length cannot exceed 254.
[error] class x43716_outr_UnitPipe_kernel(
[error]                                  ^
[error] Could not write class accel/x43580_kernel because it exceeds JVM code size limits. Method kernel's code too large!
[error] Could not write class accel/x43580_kernel$x43580_module$1 because it exceeds JVM code size limits. Method <init>'s code too large!
[error] Could not write class accel/x43710_outr_Foreach_kernel because it exceeds JVM code size limits. Method kernel's code too large!
[error] Could not write class accel/x43711_outr_Foreach_kernel because it exceeds JVM code size limits. Method kernel's code too large!
[error] Could not write class accel/x43716_outr_UnitPipe_kernel because it exceeds JVM code size limits. Method kernel's code too large!
[error] 9 errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 114 s, completed Jan 23, 2019 4:02:56 PM
make: *** [hw] Error 1

I took a look at the generated chisel. It seems that for  this particular unit pipe, the parameter size gets too huge and breaks the JVM code size. Is there a way to confine the number of parameters passed to a certain pipe? 

The app that caused the break can be found on tucson. It's at /home/tianzhao/spatial/apps/src/AFGEMM.scala.

mattfel1, 2019-02-11:
Things mentioned in this thread should all be fixed now.

closed 2019-02-11 by mattfel1

kelayamatoz, 2019-02-11:
For the record, the generated verilog file is attached to this ticket. It should take ~2hrs to synth + impl with the default number of workers.
[afgemm_16_16_srcs.tar.gz](https://github.com/stanford-ppl/spatial/files/2853255/afgemm_16_16_srcs.tar.gz)