stanford-ppl/spatial #179: bugged sm_RootController.scala

hzhang86, 2019-01-12:
Hi,

I'm testing the Spatial (master) with the InOutArg example (converted from the example in spatial-lang). It failed when I do "make aws-F1-afi" with following error:

[error] /home/centos/spatial/gen/InOutArgTest_hw/chisel/sm_RootController.scala:47:1: '{' expected.
[error] import fringe.templates._
[error] ^
[error] /home/centos/spatial/gen/InOutArgTest_hw/chisel/sm_RootController.scala:92:1: '{' expected.
[error] import fringe.templates._
[error] ^
[error] /home/centos/spatial/gen/InOutArgTest_hw/chisel/sm_RootController.scala:137:1: '{' expected.
[error] import fringe.templates._
[error] ^
[error] /home/centos/spatial/gen/InOutArgTest_hw/chisel/sm_RootController.scala:182:1: '{' expected.
[error] import fringe.templates._

Basically, the current sm_RootController.scala file has multiple definitions of the same-named object "RootController" for each iteration (I have 5 iterations in InOutArg, it generates 5 definitions of RootController with the same package and import info), which is weird. Could you check it why it's generated like that?

Thanks

mattfel1, 2019-01-12:
That seems strange.  By 5 iterations of InOutArg, do you mean you have the Accel in the app wrapped in a loop, or you have 5 Accels, or something else?

hzhang86, 2019-01-12:
Like this:

For (i <- 0 to 4) {
   Accel { }
   ...
}

hzhang86, 2019-01-15:
I also found duplicated code for the Accel block in both generated chisel file: Controllers.scala and host file: TopHost.cpp. I suspect it might be something to do with the loop unrolling optimization? The issue was not encountered in old Spatial. The example was ported from the old Spatial: UnitTests.scala/InOutArg

Thanks

mattfel1, 2019-01-18:
Hi. I just fixed an issue with multiple-uses of the Accel.  I got a different issue than the one you described, and the fix is on develop branch now.  Here is the app I used for reference:

```
@spatial class HostIOLoop extends SpatialTest {
  override def runtimeArgs: Args = "7"

  def main(args: Array[String]): Unit = {
    // Declare SW-HW interface vals
    val y = HostIO[Int]
    val N = args(0).to[Int]

    // Connect SW vals to HW vals
    setArg(y, N)

    // Create HW accelerator
    for (i <- 0 until 4) {
      Accel {
        y := y + 4
      }
    }

    val result = getArg(y)

    val gold = N + 4*4
    println("expected: " + gold)
    println("result: " + result)

    val cksum = gold == result
    println("PASS: " + cksum + " (InOutArg)")
    assert(cksum)
  }
}
```
Let me know if you still see issues with what you are working on and I will continue fixing

hzhang86, 2019-01-18:
So which branch should I use? the develop or the master?

mattfel1, 2019-01-25:
Hi.  I'll try to figure out whats happening here.  Would you be able to either send the app you are using, or send the gen/InOutArgTest_hw + logs/InOutArgTest_hw directories and I can try to figure out whats happening?  There must be something going on in your app that I didn't capture in the app above.

dkoeplin, 2019-01-25:
If he's using the `0 to 4` syntax, that's not a staged range right now, meaning he'd actually be meta-programming multiple Accel instances. Could be the cause of the duplication.

hzhang86, 2019-01-25:
Thanks, David.

That's the cause!
However, I did 0 to 4 instead of 0 until 5 because of this post:
https://spatial-lang.org/metaprog/   The current workaround is to avoid
using x until y in favor of x to (y - 1).
Honestly, that post is confusing. I still don't quite understand what
syntax is allowed inside and outside the "Accel" block, respectively, and
what syntax we use will cause an unexpected logic error because of this
"Staging" technique after reading it several times. I don't have the deep
knowledge of compiler techniques.

dkoeplin, 2019-01-26:
Yeah that's understandable, meta-programming in general is a bit hard to wrap one's head around. In this case it's doubly so because we haven't defined it terribly well. 

Basically, the problem from the language perspective is deciding, when a user writes a loop with `0 to 4` if they mean to duplicate the body of the loop or if they just want to iterate (at runtime) 4 times. We've made the rather arbitrary and somewhat non-intuitive choice that `0 until 5` means to iterate at runtime, while `0 to 4` means to duplicate. (The post is about when we _intend_ to duplicate, so the workaround is saying to use the `to` syntax if that's your intention.) Unfortunately this is a problem which stems from Spatial being embedded in Scala, and is a bit difficult to fix.

closed 2019-02-11 by mattfel1