stanford-ppl/spatial #197: Do not fuse FMA if it pulls a multiply from outside a cycle to inside a cycle

mattfel1, 2019-04-05:
For example, do not fuse in this case:
```
Foreach(N by 1){i => 
  val temp = i * i
  sram(i) = sram(i) + temp
}
```
The cycle should be SRAMRead and FixAdd.  FuseFMA comes along and fuses the mult and the add so that the cycle is now much longer (SRAMRead and FixFMA).

dkoeplin, 2019-04-05:
Just to be pedantic, this case is actually ok since the in-place update to `sram` doesn't have a required initiation interval. If you were doing register update or something like `sram(i) = sram(i-1)` though this is absolutely an issue.

mattfel1, 2019-04-05:
`sram(i) = f(sram(i-1))` is definitely a clearer example of where we would want to turn off this fma fusion.

Usually, `sram(i) = f(sram(i))` can have II = 1, but it is problematic if N is not known at compile time (i.e. ArgIn) or if there is a loop that accumulates into a memory but one of the inner-more iterators is not used in the address (i.e. `Foreach(N by 1, M by 1, P by 1){(i,j,k) => s(i,k) = s(i,k) + j`).  Setting II = 1 for `s(i) = f(s(i))` has been correct in all of our test cases but is wrong if the addressing repeats in fewer steps than the latency of the cycle in the IR.  Recently there have been some apps that exposed these kinds of issues because they parallelized the iterators by a lot or loops were fused and reordered and accumulated into lower-dimension memories.  

Either way, I reworked the iteration diff analyzer a few weeks ago to account for all these kinds of problems and it now will give you a guaranteed-correct II along with a warning that explains why its using a conservative II and how you can fix/override it.  I don't think I made an issue to tag to the fix though.

dkoeplin, 2019-04-05:
Ah interesting, I see your point. Are you running the FMA transformation before or after iteration diff analysis now? If after, it seems like you could just quickly check to see if the add you're fusing with is part of a cycle with a required initiation interval?

mattfel1, 2019-04-05:
Right now it does retiming analysis -> ... -> iter diff analysis -> ... -> unrolling -> ... -> fma transformations.  If cycle metadata is still there and correct after unrolling then I'll just check that way.  Otherwise I'll just put another retime analysis before the fma transformations.  (I separated the analysis from the transformer for retiming)

mattfel1, 2019-04-07:
Should be good now

closed 2019-04-07 by mattfel1