stanford-ppl/spatial #201: Use sbt to run app fails

kumasento, 2019-04-10:
Hi,

This issue is possibly due to my misconfiguration, but when I was trying to run a `SpatialApp` in this codebase, e.g., `HelloSpatial`, through `sbt` there was an exception raised:

```log
projects/spatial ‹master› » sbt
[info] Loading settings from idea.sbt ...
[info] Loading global plugins from /home/ruizhe/.sbt/1.0/plugins
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/ruizhe/projects/spatial/project
[info] Updating ProjectRef(uri("file:/home/ruizhe/projects/spatial/project/"), "spatial-build")...
[info] Done updating.
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/home/ruizhe/projects/spatial/)
[info] sbt server started at local:///home/ruizhe/.sbt/1.0/server/552e58e58d0e437d3363/sock
sbt:spatial> project apps
[info] Set current project to apps (in build file:/home/ruizhe/projects/spatial/)
sbt:apps> runMain HelloSpatial --sim
[info] Updating utils...
[info] Updating emul...
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- org.scala-lang:scala-compiler:2.12.6               (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- org.scala-lang:scala-compiler:2.12.6               (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:emul_2.12:1.1-SNAPSHOT        (depends on 1.0.6)
[warn] 	    +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Updating forge...
[info] Updating models...
[info] Updating poly...
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- org.scala-lang:scala-compiler:2.12.6               (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:forge_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- org.scala-lang:scala-compiler:2.12.6               (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:poly_2.12:1.1-SNAPSHOT        (depends on 1.0.6)
[warn] 	    +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- edu.stanford.cs.dawn:models_2.12:1.1-SNAPSHOT      (depends on 1.0.6)
[warn] 	    +- org.scala-lang:scala-compiler:2.12.6               (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Updating argon...
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- edu.stanford.cs.dawn:forge_2.12:1.1-SNAPSHOT       (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:emul_2.12:1.1-SNAPSHOT        (depends on 1.1.0)
[warn] 	    +- org.scala-lang:scala-compiler:2.12.6               (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:argon_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[... 82 lines cut ...]
[warn] 3 warnings found
[failed] Total time: 3.6450 seconds

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error] 	at sbt.Run$.executeTrapExit(Run.scala:124)
[error] 	at sbt.Run.run(Run.scala:77)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error] 	at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.Try$.apply(Try.scala:209)
[error] 	at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 139 s, completed 10-Apr-2019 17:21:36
```

But after I run `make apps`, everything turns out to be OK.

Is this an expected behaviour?

Thanks

mattfel1, 2019-04-10:
Can you try running again with --vv flag?  Also can you double check that it was able to compile emptiness properly?  Try `cd spatial/poly/resources && gcc emptiness.c -lisl`

dkoeplin, 2019-04-10:
Just throwing this out there, `make apps` includes a `make install` call which builds the ISL dependency. Possibly it's not catching the missing dependency properly there in a nice way.

mattfel1, 2019-04-10:
We actually moved the emptiness compilation to inside the ISL trait so that when we deployed the jars for quickstart, the emptiness executable would get created during normal app compilation and be sensitive to the version of the executable (emptiness.c has changed once so far).  My guess is that the error here happens during that compilation

kumasento, 2019-04-10:
Thanks for all your reply. Here I've collected more information based on your suggestions (including commands for reproduction):

```shell
git clone https://github.com/stanford-ppl/spatial
cd spatial
sbt "; project apps; runMain HelloSpatial --vv --sim"
```

Commit ID: `8cae8e4f22b25348561949f066e4840d80066fcd`

Output:

```log
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /mnt/ccnas2/bdp/rz3515/projects/sandbox/spatial/project
[info] Updating ProjectRef(uri("file:/mnt/ccnas2/bdp/rz3515/projects/sandbox/spatial/project/"), "spatial-build")...
[info] Done updating.
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/mnt/ccnas2/bdp/rz3515/projects/sandbox/spatial/)
[info] Set current project to apps (in build file:/mnt/ccnas2/bdp/rz3515/projects/sandbox/spatial/)
[info] Updating emul...
[info] Updating utils...
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn]  * org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn]      +- edu.stanford.cs.dawn:emul_2.12:1.1-SNAPSHOT        (depends on 1.1.0)
[warn]      +- org.scala-lang:scala-compiler:2.12.6               (depends on 1.0.6)
[warn]      +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn]  * org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn]      +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.1.0)
[warn]      +- org.scala-lang:scala-compiler:2.12.6               (depends on 1.0.6)
[warn]      +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Updating poly...
[info] Updating forge...
[info] Updating models...
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn]  * org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn]      +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.1.0)
[warn]      +- edu.stanford.cs.dawn:poly_2.12:1.1-SNAPSHOT        (depends on 1.1.0)
[warn]      +- org.scala-lang:scala-compiler:2.12.6               (depends on 1.0.6)
[warn]      +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn]  * org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn]      +- edu.stanford.cs.dawn:models_2.12:1.1-SNAPSHOT      (depends on 1.1.0)
[warn]      +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.1.0)
[warn]      +- org.scala-lang:scala-compiler:2.12.6               (depends on 1.0.6)
[warn]      +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Done updating.
[spatial] 0:..ndbox/spatial*                                                    
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn]  * org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn]      +- edu.stanford.cs.dawn:forge_2.12:1.1-SNAPSHOT       (depends on 1.1.0)
[warn]      +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.1.0)
[warn]      +- org.scala-lang:scala-compiler:2.12.6               (depends on 1.0.6)
[warn]      +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Updating argon...
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn]  * org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn]      +- edu.stanford.cs.dawn:forge_2.12:1.1-SNAPSHOT       (depends on 1.1.0)
[warn]      +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.1.0)
[... 177 lines cut ...]
[bug]   argon.codegen.StructCodegen.postprocess$(StructCodegen.scala:22)
[bug]  .. [see /mnt/ccnas2/bdp/rz3515/projects/sandbox/spatial/logs/HelloSpatial/HelloSpatial_exception.log]
[bug] This is due to a compiler bug. A log file has been created at:
[bug]   /mnt/ccnas2/bdp/rz3515/projects/sandbox/spatial/logs/HelloSpatial/HelloSpatial_exception.log
[warn] 3 warnings found
[failed] Total time: 6.4360 seconds

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error]         at sbt.Run$.executeTrapExit(Run.scala:124)
[error]         at sbt.Run.run(Run.scala:77)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error]         at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at scala.util.Try$.apply(Try.scala:209)
[error]         at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error]         at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error]         at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 155 s, completed 10-Apr-2019 22:20:03
```

And compiling the `emptiness.c` file results in the same error.

dkoeplin, 2019-04-10:
@mattfel1 ah, but we don't yet build `files_list` automatically :P

mattfel1, 2019-04-10:
Heh, yep.  I'll make that happen automatically.

@kumasento try running bin/update_resources.sh and then recompile.  It looks like there are some missing dependencies that should be fixed by this script

kumasento, 2019-04-11:
@mattfel1 Yes, it fixes the issue, thanks!

closed 2019-04-11 by kumasento