stanford-ppl/spatial #203: Adding new Transformer breaks compilation

kumasento, 2019-04-11:
Hi, I was trying to add a new `Transformer` to `spatial` that actually does nothing (for experimental reasons) and the code looks as follows:

```scala
case class DummyTransformer(IR: State) extends MutateTransformer with AccelTraversal {

  override def transform[A: Type](lhs: Sym[A], rhs: Op[A])(implicit ctx: SrcCtx): Sym[A] = rhs match {
    case _: AccelScope => inAccel {
      super.transform(lhs, rhs)
    }
  }
}
```

But when I was building the codebase through `sbt`, I got a huge amount of error messages produced:

```log
[info] Compiling 2 Scala sources to /home/ruizhe/projects/spatial/target/scala-2.12/classes ...
[error] ## Exception when compiling 334 sources to /home/ruizhe/projects/spatial/target/scala-2.12/classes
[error] null
[error] scala.tools.nsc.transform.ExplicitOuter$ExplicitOuterTransformer.transform(ExplicitOuter.scala:362)
[error] scala.tools.nsc.transform.ExplicitOuter$ExplicitOuterTransformer.transform(ExplicitOuter.scala:323)
[error] scala.reflect.internal.Trees.itransform(Trees.scala:1366)
[error] scala.reflect.internal.Trees.itransform$(Trees.scala:1357)
[error] scala.reflect.internal.SymbolTable.itransform(SymbolTable.scala:18)
[error] scala.reflect.internal.SymbolTable.itransform(SymbolTable.scala:18)
[error] scala.reflect.api.Trees$Transformer.transform(Trees.scala:2555)
[error] scala.tools.nsc.transform.TypingTransformers$TypingTransformer.transform(TypingTransformers.scala:44)
[error] scala.tools.nsc.transform.ExplicitOuter$OuterPathTransformer.scala$reflect$internal$Trees$UnderConstructionTransformer$$super$transform(ExplicitOuter.scala:205)
[error] scala.reflect.internal.Trees$UnderConstructionTransformer.transform(Trees.scala:1720)
....
[error] 	at scala.reflect.api.Trees$Transformer.transform(Trees.scala:2555)
[error] 	at scala.tools.nsc.transform.TypingTransformers$TypingTransformer.transform(TypingTransformers.scala:44)
[error] 	at scala.tools.nsc.transform.ExplicitOuter$OuterPathTransformer.scala$reflect$internal$Trees$UnderConstructionTransformer$$super$transform(ExplicitOuter.scala:205)
[error] 	at scala.reflect.internal.Trees$UnderConstructionTransformer.transform(Trees.scala:1720)
[error] 	at scala.reflect.internal.Trees$UnderConstructionTransformer.transform$(Trees.scala:1715)
[error] 	at scala.tools.nsc.transform.ExplicitOuter$OuterPathTransformer.transform(ExplicitOuter.scala:282)
[IJ]sbt:spatial> [error] (Compile / compileIncremental) java.lang.StackOverflowError
[error] Total time: 7 s, completed 11-Apr-2019 17:10:42
```

I tried to clean up caches by `sbt clean`, as well as increasing the heap size to 8192 MB, this issue persists.

Could anyone kindly give me a hint about the possible reason? Thanks!

dkoeplin, 2019-04-11:
Wow. Looks like you hit a scalac typing bug right out of the gate, sorry about that. Could you try commenting out this new class just to verify that's the cause? 

I don't think this is causing the issue, but just for future reference, you should add a 
`case _ => super.transform(lhs, rhs)` as a wildcard case to your match block to avoid a match exception at runtime (otherwise it won't know what to do with things that aren't AccelScope).

kumasento, 2019-04-12:
Hi @dkoeplin Thank you for your reply!

I have created a new branch in my fork that includes this new transformer (with the revision suggested by you) at https://github.com/kumasento/spatial/tree/test-add-trans. I tried to comment out the initialization of the `DummyTransformer` while the compilation still cannot pass.

> otherwise it won't know what to do with things that aren't AccelScope

Yes indeed, I didn't notice the `AccelScope` after the wildcard when I was copying this snippet from another `Transformer` :(

Please let me know if there is anything else I could do

dkoeplin, 2019-04-12:
Cool, I'll take a look to see if I can reproduce the problem

dkoeplin, 2019-04-12:
Taking a look at what you posted again, the error you're seeing is during scala compilation itself, not while running, and based on what you pasted it looks like a stack overflow error during the scala compiler's typing pass. Naturally, increasing the heap size won't help solve a stack overflow. Could you check what your stack size is and increase it? (I use -Xss1G but that's probably more than necessary)

kumasento, 2019-04-12:
Ah it is working now, thanks and sorry for this silly mistake!

(PS I attached `-J-Xss1G` to `sbt` when running the compilation)

closed 2019-04-12 by kumasento