stanford-ppl/spatial #205: Help Wanted for Creating Early Area Estimate Based on the Controller Tree

kelayamatoz, 2019-04-11:
The current approach to get an area estimate is by running through the synthesis tool chain. This approach doesn't scale well due to the overhead generated by chisel generation, firrtl compilation, and fpga synthesis. It seems that the controller tree should provide an rough estimation about the memory, logic, and register utilization. If we can get this estimation before hitting the hard FPGA / RTL generation route, that would be great!

dkoeplin, 2019-04-11:
[https://github.com/stanford-ppl/spatial/blob/master/src/spatial/traversal/AreaAnalyzer.scala](https://github.com/stanford-ppl/spatial/blob/master/src/spatial/traversal/AreaAnalyzer.scala) ?

mattfel1, 2019-04-11:
We will definitely start from there.  I think area estimation pre-unrolling/mem analysis is going to be tricky now since small changes in parallelizations and dse parameters can have huge impacts on luts and brams that will be instantiated, and it will make dse take way longer to put mem analysis in the loop.  This issue is asking for area estimation only based on the last IR so that we can cut out synthesis and p&r from the designer's development flow, and we can think about fully automated area estimation in dse at some point in the future.

Also we need to collect new area data to train on