stanford-ppl/spatial #211: Vectorize inner loop access

yaqiz01, 2019-04-18:
Example where inner loop accesses using different memories as supposed be vectorized.
```
@spatial abstract class VecTest extends SpatialTest { self =>

  type X = Float

  val N:scala.Int = 32
  val op:scala.Int = 2
  val ip:scala.Int = 8

  val margin = 1

  def vectest[T: Num](vector: Array[T]) = {

    val vDRAM = DRAM[T](N)
    val mDRAM = DRAM[T](N, N)
    setMem(vDRAM, vector)

    Accel {
      val vTile = SRAM[T](N)
      vTile load vDRAM(0 :: N par ip) // Load mu0
      val mTile = SRAM[T](N, N)
      Foreach(N by 1 par op) { ii =>
        Foreach(N par ip) { jj =>
          mTile(ii, jj) = vTile(ii) * vTile(jj)
        }
      }
      mDRAM(0 :: N, 0 :: N par ip) store mTile
    }

    getMem(mDRAM)
  }

  def main(args: Array[String]): Unit = {
    val vector = Array.tabulate(N) { i => i }
    val result = vectest(vector)

    val gold = Array.tabulate(N) { i => Array.tabulate(N) { j => vector(i) * vector(j) } }.flatten

    printArray(gold, "gold: ")
    printArray(result, "result: ")

    val cksum = gold.zip(result){(a,b) => abs(a-b) <= margin}.reduce{_&&_}
    println("PASS: " + cksum  + " (VecTest)")
    assert(cksum)
  }

}
```

mattfel1, 2019-04-20:
Really tiny fix but vectorizing should be much more human-readable now.  

The issue was that when grouping accesses, it was basically only sorting by uid so different accesses got interleaved.  I think interleaving the accesses caused it to come up with really confusing (but functionally correct 🙃 ) access groups.  Now I just keep accesses together when iterating through and trying to group.  It will work in the cases mentioned above, but I'm not exactly sure if it will vectorize nicely with some other complicated patterns (i.e. accesses uses iterators that have more uid lanes inside of it), but I'm pretty confident it will at least be more readable than it has been before.

mattfel1, 2019-04-20:
2bee1406e54cb0ee416cffc218eb418504543554 for reference

closed 2019-04-20 by mattfel1