stanford-ppl/spatial #229: Iteration Diff Break with new access grouping scheme

yaqiz01, 2019-05-09:
Now with `--bank-groupUnroll` flag will try to group accesses with the same sym first before trying to group across syms in memory configurer. This flag currently breaks IterationDiff regression test.

mattfel1, 2019-05-16:
the thing it fails on is 
```
/* 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 20  // II = single lane latency * 1, has double segmentation
       *   O   O   O   O      
       *   |_X_|   |  --->
       *       |_X_|
       */
```
My guess is that you vectorize this access, whereas the iteration diff analyzer is setting metadata that says you must split it into 2 nodes and you need to start the second read after the first write finishes
sorry, the opposite of what I just said
the access pattern looks like
```
'FREACH2.Foreach(start until stop by step par 2){ i => 
          val a = sram(i)
          val b = sram(i+1)
          val lower = mux(a < b, a, b)
          val upper = mux(a < b, b, a)
          sram(i) = lower
          sram(i+1) = upper
        }
```

its pairwise sorting two adjacent elements, and that is par 2 so it has to finish the first sort before it starts the second.  So the reads and writes are separate nodes.  I assume your flag doesn't account for any cases that are this stupid?
maybe we should just detect this kind of thing and crash?

yaqiz01, 2019-05-16:
Added error message on fda46f6

closed 2019-05-16 by yaqiz01