stanford-ppl/spatial #255: FloatingPoint Class Doesn't Have toSeq Method

kelayamatoz, 2019-06-26:
Our chiselgen runs toSeq over a Wire object. However, fringe.templates.math.FloatingPoint class doesn't have a toSeq method. I'm testing a patch that adds a toSeq method to the FloatingPoint class.

kelayamatoz, 2019-06-26:
Added patch to branch issue255_fltpt2seq. 
@mattfel1 Should I merge it into develop or wait for the admin to merge? 
Test app attached:  [HelloFloat.scala.zip](https://github.com/stanford-ppl/spatial/files/3332144/HelloFloat.scala.zip)

mattfel1, 2019-06-26:
This one line looks safe enough so you can merge. Are you working on floating point implementations?

kelayamatoz, 2019-06-26:
I had an app that required high-precision non-linear function implementations for FPGA and that's why I was hitting this issue. I'm not implementing floating point for everything.

By looking at the fringe templates, it seems that the floating point (32 bit) implementations are there already. Is there still anything missing?

closed 2019-06-26 by kelayamatoz

mattfel1, 2019-06-26:
Anything floating point is use-at-your-own risk right now.  Last I checked, only addition, subtraction, and multiplication are semi-trustworthy

kelayamatoz, 2019-06-26:
Got it. I will see how far I can get.