stanford-ppl/spatial #263: Accum register gets dropped during transient cleanup

kelayamatoz, 2019-08-15:
In the attached app, the accumulated register gets dropped during use analysis. Adding a println prevents it from being dropped in scala sim. 

[AccumType15.scala.zip](https://github.com/stanford-ppl/spatial/files/3507207/AccumType15.scala.zip)

mattfel1, 2019-08-15:
One issue is retiming nodes that read from a specialized accumulator:
```
              x218 = RegNew(Const(0))
              x238 = RegRead(x218)
               - class spatial.metadata.retiming.FullDelay: FullDelay(0.0)
              x241 = SRAMBankedWrite(x158,Vector(x238),Vector(List(Const(0))),Vector(b214),Vector(Set(x236)))
               - class spatial.metadata.retiming.FullDelay: FullDelay(0.2)
              x302 = RegAccumOp(x218,x230,Set(x315, x316),AccumAdd,x314)
               - class spatial.metadata.retiming.FullDelay: FullDelay(9.2)
```
Second issue seems to be the FixOpAccum chisel node ignores the "first" signal and keeps a running count no matter what

kelayamatoz, 2019-08-19:
I'm looking at AccumReplacements.scala, test 13. If we have a case where the accum does not use the inner most counter,  e.g.
```scala
val acc = Reg[T]
Foreach (2 by 1, 4 by 1, 2 by 1) { (i, j, k) => 
  val update = ...
  acc := mux(i == 0 && j == 0, t, t + acc.value)
}
```
Should the acc be marked as a specialized accum? If not, do we need to match this pattern on any template?

mattfel1, 2019-08-19:
What you have should not be marked since the iterator k is more inside of the iterators used in the the condition.  

The test 13 that I have is based on mux(j == 0 && k == 0,...).  Not sure if its something I changed later or you changed.  This one should be replaced though

kelayamatoz, 2019-08-19:
I manage to handle the original case in test 13; I'm trying to come up with new cases that have similar patterns to test 13 but should / shouldn't work.

closed 2019-08-21 by kelayamatoz