stanford-ppl/spatial #269: FIFO not duplicated during outer loop unrolling

yaqiz01, 2019-09-20:
FIFO read by the unrolled outer loop is not duplicated in the IR.
Test FIFOParRead

yaqiz01, 2019-09-20:
Branch issue269_fifo

mattfel1, 2019-09-21:
I'm looking at it now.  I think the IR is actually correct but the chisel template is messing up.  It banks the FIFO by 2, which is correct since the two unrolled deqs may conflict and it can serve both of them with this banking scheme.    The vcs sim gives all 0s now so I'm looking into that.

mattfel1, 2019-09-22:
This is confusing.  In the template, I assume that a FIFO needs to serve as many elements as its widest reader.  This means that outer loop unrolling of a read, or multiple reads in the same inner controller, are not supported properly in the template even though they are handled correctly during banking.  The information about concurrency of the different accesses is lost by the time we reach the chisel template.

I'm going to make the FIFO template a little more complicated by allowing it serve all readers simultaneously, regardless of their concurrency.  I don't think this will have too big of an impact since most apps use FIFOs as a single-writer, single-reader type of queue.

yaqiz01, 2019-09-22:
It's actually not a feature I need. I just want to make sure I interpret the IR correctly. If it's too much trouble we can leave it for now

mattfel1, 2019-09-23:
Fixed.  Multiple enqs and deqs can operate simultaneously now as long as the banking analyzer banked properly for them.

closed 2019-09-23 by mattfel1