stanford-ppl/spatial #278: cl_mstr_axi_tst.sv not exists in version 1.4.7 of Amazon's EC2 FPGA Hardware and Software Development Kit

tissue3, 2019-10-28:
I tried to follow the instructions on https://github.com/stanford-ppl/spatial-multiverse/blob/master/docs/aws.md to generate bitstream on AWS F1 and got the following error.
```
## file copy -force $CL_DIR/design/cl_mstr_axi_tst.sv                 $TARGET_DIR
error copying "/home/centos/spatials/aws-fpga//hdk/cl/examples/GEMM_NCubed/design/cl_mstr_axi_tst.sv": no such file or directory
    while executing
"file copy -force $CL_DIR/design/cl_mstr_axi_tst.sv                 $TARGET_DIR"
    (file "encrypt.tcl" line 47)
```
This is strange, because `cl_mstr_axi_tst.sv` does not exist in version 1.4.7 of [Amazon's EC2 FPGA Hardware and Software Development Kit](https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples/cl_dram_dma/design) and I am following the instructions saying "Spatial was most recently tested with version 1.4.7 of this repository".

shadjis, 2019-10-30:
For me, I can only ensure that AWS F1 works with the version of Spatial in the Spatial Multiverse:
```
https://github.com/stanford-ppl/spatial/tree/c1972756e7a1300b5c2870876059c3e0aaa391c7
```

Any other version of Spatial first needs to merge in my changes on that branch.

For example, notice these changes to `encrypt.tcl` that I made on July 26:
```
https://github.com/stanford-ppl/spatial/commit/e2a6a98ed4175d6503d42491def7398cf3ae1ce2#diff-8226eec3faff0e6aa18675efc370d733
```

Among these changes is the deletion of:
```
file copy -force $CL_DIR/design/cl_mstr_axi_tst.sv
```

If we now look at Spatial's master branch, we do not see these changes in encrypt.tcl:

**Incorrect:**

```
https://github.com/stanford-ppl/spatial/blob/master/resources/synth/aws.sw-resources/build/encrypt.tcl
```
```
https://github.com/stanford-ppl/spatial/blob/develop/resources/synth/aws.sw-resources/build/encrypt.tcl
```

**Correct:**

```
https://github.com/stanford-ppl/spatial/blob/c1972756e7a1300b5c2870876059c3e0aaa391c7/resources/synth/aws.sw-resources/build/encrypt.tcl
```

shadjis, 2019-12-01:
hi @mattfel1 
I think you have now merged these branches into master so this issue can be closed

closed 2019-12-02 by mattfel1