stanford-ppl/spatial #279: Compilation error with Zynq fpag

tissue3, 2019-10-29:
Following the advice in #277 , I tried to use Zynq as target device. However, this result in errors at compilation stage.
I used spatial-quickstart run the following commands:
```
bin/spatial HelloSpatial --synth --fpga=Zynq
cd gen/HelloSpatial
sbt "compile"
```
This results in a lot of errors.
```
[info] Compiling 42 Scala sources to /home/zl679/Desktop/spatial-quickstart/gen/HelloSpatial/target/scala-2.12/classes ...
[info] Non-compiled module 'compiler-bridge_2.12' for Scala 2.12.6. Compiling...
[info]   Compilation completed in 11.416s.
[error] /home/zl679/Desktop/spatial-quickstart/gen/HelloSpatial/chisel/AccelTop.scala:24:11: not found: type AbstractAccelTop
[error] ) extends AbstractAccelTop with IOModule { 
[error]           ^
[error] /home/zl679/Desktop/spatial-quickstart/gen/HelloSpatial/chisel/IOModule.scala:74:42: not found: type CXPAccelInterface
[error]     case _:targets.cxp.CXP     => IO(new CXPAccelInterface(io_w, io_v, globals.LOAD_STREAMS, globals.STORE_STREAMS, globals.GATHER_STREAMS, globals.SCATTER_STREAMS, globals.numAllocators, io_numArgIns, io_numArgOuts))
[error]                                          ^
[error] /home/zl679/Desktop/spatial-quickstart/gen/HelloSpatial/chisel/IOModule.scala:75:18: not enough arguments for constructor CustomAccelInterface: (io_w: Int, io_v: Int, io_loadStreamInfo: List[fringe.StreamParInfo], io_storeStreamInfo: List[fringe.StreamParInfo], io_gatherStreamInfo: List[fringe.StreamParInfo], io_scatterStreamInfo: List[fringe.StreamParInfo], io_axiStreamsIn: List[fringe.templates.axi4.AXI4StreamParameters], io_axiStreamsOut: List[fringe.templates.axi4.AXI4StreamParameters], io_numAllocators: Int, io_numArgIns: Int, io_numArgOuts: Int)fringe.CustomAccelInterface.
[error] Unspecified value parameters io_numArgIns, io_numArgOuts.
[error]     case _ => IO(new CustomAccelInterface(io_w, io_v, globals.LOAD_STREAMS, globals.STORE_STREAMS, globals.GATHER_STREAMS, globals.SCATTER_STREAMS, globals.numAllocators, io_numArgIns, io_numArgOuts))
[error]                  ^
[error] /home/zl679/Desktop/spatial-quickstart/gen/HelloSpatial/chisel/IOModule.scala:70:11: value streamInsInfo is not a member of object fringe.globals
[error]   globals.streamInsInfo = io_streamInsInfo
[error]           ^
[error] /home/zl679/Desktop/spatial-quickstart/gen/HelloSpatial/chisel/IOModule.scala:71:11: value streamOutsInfo is not a member of object fringe.globals
[error]   globals.streamOutsInfo = io_streamOutsInfo
[error]           ^
[error] /home/zl679/Desktop/spatial-quickstart/gen/HelloSpatial/chisel/Instantiator.scala:14:24: not found: type Top
[error] class TopUnitTester(c: Top)(implicit args: Array[String]) extends ArgsTester(c) {
[error]                        ^
[error] /home/zl679/Desktop/spatial-quickstart/gen/HelloSpatial/chisel/Instantiator.scala:14:78: no arguments allowed for nullary constructor Object: ()Object
[error] class TopUnitTester(c: Top)(implicit args: Array[String]) extends ArgsTester(c) {
[error]                                                                              ^
[error] /home/zl679/Desktop/spatial-quickstart/gen/HelloSpatial/chisel/Instantiator.scala:18:18: not found: type Top
[error]   type DUTType = Top
[error]                  ^
[error] /home/zl679/Desktop/spatial-quickstart/gen/HelloSpatial/chisel/Instantiator.scala:56:9: not found: type Top
[error]     new Top(this.target, () => Module(new AccelTop(w, numArgIns, numArgOuts, numArgIOs, numArgOuts_instr + numArgBreakpts, numAllocators, loadStreamInfo, storeStreamInfo, gatherStreamInfo, scatterStreamInfo, streamInsInfo, streamOutsInfo)))
[error]         ^
[error] /home/zl679/Desktop/spatial-quickstart/gen/HelloSpatial/chisel/Instantiator.scala:58:32: type mismatch;
[error]  found   : top.TopUnitTester
[error]  required: fringe.ArgsTester[top.Instantiator.DUTType]
[error]     (which expands to)  fringe.ArgsTester[<error>]
[error]   def tester = { c: DUTType => new TopUnitTester(c) }
[error]                                ^
[error] 10 errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 27 s, completed 29-Oct-2019 11:47:05 AM
```

mattfel1, 2019-10-29:
This looks like there are old packages hanging around.  Can you try deleting these directories and trying again?

~/.ivy2/cache/edu.stanford.cs.dawn
~/spatial-quickstart/target
~/spatial-quickstart/project/target
~/spatial-quickstart/gen

?

tissue3, 2019-10-29:
It works now.

closed 2019-10-29 by tissue3

mattfel1, 2019-10-29:
Just caught a mistake in the synth makefile.  I will push a fix, but in the short term you should edit zynq.hw-resources/build/Makefile.  Change line 24 from `sed -i 's/design_1_SpatialIP_0_0/Top/' design_1.v` to sed -i 's/design_1_SpatialIP_0_0/SpatialIP/' design_1.v`

tissue3, 2019-10-30:
Thanks for reminding. I manually changed it and reran the flow. Everything works fine till Phase 3 of vivado synthesis. "The packing of instances into the device could not be obeyed." Any insight that why this would occur? I am using Vivado v2017.2_sdx.
```
Phase 4 Add Constraints new method
Phase 4.1 Build CellView Core
Phase 4.1 Build CellView Core | Checksum: 418df461

Time (s): cpu = 00:06:24 ; elapsed = 00:03:31 . Memory (MB): peak = 3162.660 ; gain = 33.109 ; free physical = 4739 ; free virtual = 49878

Phase 4.2 Add User PBlocks
Phase 4.2 Add User PBlocks | Checksum: 418df461

Time (s): cpu = 00:06:24 ; elapsed = 00:03:31 . Memory (MB): peak = 3162.660 ; gain = 33.109 ; free physical = 4739 ; free virtual = 49878

Phase 4.3 Apply User PBlocks
Phase 4.3 Apply User PBlocks | Checksum: 418df461

Time (s): cpu = 00:06:24 ; elapsed = 00:03:31 . Memory (MB): peak = 3162.660 ; gain = 33.109 ; free physical = 4739 ; free virtual = 49878

Phase 4.4 Add terminal Constraints
Phase 4.4 Add terminal Constraints | Checksum: 418df461

Time (s): cpu = 00:06:24 ; elapsed = 00:03:31 . Memory (MB): peak = 3162.660 ; gain = 33.109 ; free physical = 4739 ; free virtual = 49878
Phase 4 Add Constraints new method | Checksum: 418df461

Time (s): cpu = 00:06:24 ; elapsed = 00:03:31 . Memory (MB): peak = 3162.660 ; gain = 33.109 ; free physical = 4739 ; free virtual = 49878
Ending Placer Task | Checksum: b5354623

Time (s): cpu = 00:06:25 ; elapsed = 00:03:31 . Memory (MB): peak = 3162.660 ; gain = 33.109 ; free physical = 4838 ; free virtual = 49977
42 Infos, 1 Warnings, 0 Critical Warnings and 3 Errors encountered.
place_design failed
ERROR: [Common 17-69] Command failed: Placer could not place all instances
INFO: [Common 17-206] Exiting Vivado at Tue Oct 29 18:10:35 2019...
```

tissue3, 2019-10-30:
I parse it as a csv and sort by LUT usage and got this:
```
(top) | 55186(103.73%)
-- | --
design_1 | 55186(103.73%)
SpatialIP | 38967(73.25%)
FringeZynq | 33842(63.61%)
Fringe | 33428(62.83%)
DRAMArbiter_1_817 | 15176(28.53%)
StreamControllerGather | 13878(26.09%)
GatherBuffer | 13256(24.92%)
FIFO_61 | 13256(24.92%)
FFRAM_1 | 13058(24.55%)
DRAMArbiter | 8598(16.16%)
AccelUnit | 5125(9.63%)
```
It seems no particular component used too much resources but when they addup, it becomes unacceptably large. Again, I am using `HelloSpatial` and didn't modify any line in the code. The design seems not particularly large, why would it be too large?

mattfel1, 2019-10-30:
Thanks for pointing this out.  It seems like we've never actually tried to synthesize this app before :grimacing: .  We have usually just used it to show how to do miscellaneous things in spatial, but the load+store+scatter+gather streams seem to add up to a DMA that is way too big for the Zynq (Fringe is usually around 25% of this board if I recall correctly.  62.83% is huge.  Most of the area for real apps should go into the AccelUnit).  If your goal is to just get the flow working from start to finish, you can just comment out the scatter and gather lines  (49 and 51) and I think it should fit (testing now).