stanford-ppl/spatial #282: error with compile HelloSpatial.scala

CristinaZN, 2019-11-12:
when i start with tutorial, it seems stuck here. 

`spatial$ bin/spatial HelloSpatial --synth --fpga=VCS
[info] Loading global plugins from /home/cristinazn/.sbt/1.0/plugins
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/cristinazn/Downloads/tslab/spatial/project
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/home/cristinazn/Downloads/tslab/spatial/)
[info] Set current project to apps (in build file:/home/cristinazn/Downloads/tslab/spatial/)
[info] Compiling 1 Scala source to /home/cristinazn/Downloads/tslab/spatial/apps/target/scala-2.12/classes ...
[info] Done compiling.
[success] Total time: 17 s, completed Nov 12, 2019 7:50:23 PM
[info] Loading global plugins from /home/cristinazn/.sbt/1.0/plugins
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/cristinazn/Downloads/tslab/spatial/project
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/home/cristinazn/Downloads/tslab/spatial/)
[info] Set current project to test (in build file:/home/cristinazn/Downloads/tslab/spatial/)
[info] Compiling 265 Scala sources to /home/cristinazn/Downloads/tslab/spatial/test/target/scala-2.12/test-classes ...
[info] /home/cristinazn/Downloads/tslab/spatial/test/spatial/tests/feature/math/FixBasics2.scala:69:9: Type annotation recommended for var declaration.
[info]     var allgood = true
[info]         ^
[info] /home/cristinazn/Downloads/tslab/spatial/test/spatial/tests/feature/math/FloatBasics2.scala:70:9: Type annotation recommended for var declaration.
[info]     var allgood = true
[info]         ^
[warn] /home/cristinazn/Downloads/tslab/spatial/test/spatial/tests/feature/streaming/FilterStream1D.scala:49:97: postfix operator !! should be enabled
[warn] by making the implicit value scala.language.postfixOps visible.
[warn] This can be achieved by adding the import clause 'import scala.language.postfixOps'
[warn] or by setting the compiler option -language:postfixOps.
[warn] See the Scaladoc for value scala.language.postfixOps for a discussion
[warn] why the feature should be explicitly enabled.
[warn]     val filename = "cat ../src/findedges.py | grep \"calibfile =\" | sed \"s/calibfile = //g\"" !!
[warn]                                                                                                 ^
[warn] /home/cristinazn/Downloads/tslab/spatial/test/spatial/tests/feature/streaming/FilterStream1D.scala:52:35: postfix operator ! should be enabled
[warn] by making the implicit value scala.language.postfixOps visible.
[warn]     "python3 ../src/findedges.py" !
[warn]                                   ^
[warn] /home/cristinazn/Downloads/tslab/spatial/test/spatial/tests/feature/streaming/FilterStream1D.scala:57:43: postfix operator ! should be enabled
[warn] by making the implicit value scala.language.postfixOps visible.
[warn]     s"echo EXAMPLE SCALA METAPROGRAMMING" !
[warn]                                           ^
`

mattfel1, 2019-11-12:
Hi.  It looks like it is just hanging while it compiles all the test apps.  If you want this to speed up, can you try just deleting everything in spatial/test/spatial/tests/* and see if it completes?  HelloSpatial is in spatial/apps/src so that app should still exist after deleting all of the tests.

CristinaZN, 2019-11-13:
i deleted all of the tests. it cannot compile emptiness.c
spatial$ bin/spatial HelloSpatial --sim
[info] Loading global plugins from /home/cristinazn/.sbt/1.0/plugins
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/cristinazn/Downloads/tslab/spatial/project
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/home/cristinazn/Downloads/tslab/spatial/)
[info] Set current project to apps (in build file:/home/cristinazn/Downloads/tslab/spatial/)
[success] Total time: 4 s, completed Nov 13, 2019 3:27:25 PM
[info] Loading global plugins from /home/cristinazn/.sbt/1.0/plugins
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/cristinazn/Downloads/tslab/spatial/project
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/home/cristinazn/Downloads/tslab/spatial/)
[info] Set current project to test (in build file:/home/cristinazn/Downloads/tslab/spatial/)
[success] Total time: 5 s, completed Nov 13, 2019 3:27:36 PM
Updating resources with bin/update_resources.sh...

**** SBT Command: ****
sbt "apps/runMain HelloSpatial --name HelloSpatial --sim"
**********************

[info] Loading global plugins from /home/cristinazn/.sbt/1.0/plugins
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/cristinazn/Downloads/tslab/spatial/project
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/home/cristinazn/Downloads/tslab/spatial/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Packaging /home/cristinazn/Downloads/tslab/spatial/target/scala-2.12/spatial_2.12-1.1-SNAPSHOT.jar ...
[info] Done packaging.
[info] Running HelloSpatial --name HelloSpatial --sim
[warn] No target specified. Specify target using: --fpga <device> or
[warn] override val target = <device>
[warn] Defaulting to 'Default' device.
[warn] Area model file Zynq_Area.csv for target Zynq was missing expected fields: 
[warn] BRAM
[info] Compiling HelloSpatial to /home/cristinazn/Downloads/tslab/spatial/gen/HelloSpatial/
[warn] libisl appears to be missing!  Please install http://isl.gforge.inria.fr/
[bug] An exception was encountered while compiling HelloSpatial: 
[bug]   Error compiling emptiness
[bug]   java.lang.Exception: Error compiling emptiness
[bug] This is due to a compiler bug. A log file has been created at: 
[bug]   /home/cristinazn/Downloads/tslab/spatial/logs/HelloSpatial//HelloSpatial_exception.log
[warn] 1 warning found
[failed] Total time: 6.9510 seconds

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error] 	at sbt.Run$.executeTrapExit(Run.scala:124)
[error] 	at sbt.Run.run(Run.scala:77)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error] 	at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.Try$.apply(Try.scala:209)
[error] 	at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (apps / Compile / runMain) Nonzero exit code: 1
[error] Total time: 13 s, completed Nov 13, 2019 3:28:26 PM
[error] Spatial compilation failed
cristinazn@cristinazn-GL553VD:~/Downloads/tslab/spatial$ cd poly/resources && gcc emptiness.c -lisl
emptiness.c:6:10: fatal error: isl/vec.h: No such file or directory
 #include <isl/vec.h>
          ^~~~~~~~~~~
compilation terminated.
cristinazn@cristinazn-GL553VD:~/Downloads/tslab/spatial/poly/resources$ ./bin/update_resources.sh 
bash: ./bin/update_resources.sh: No such file or directory
cristinazn@cristinazn-GL553VD:~/Downloads/tslab/spatial/poly/resources$ cd ../..
cristinazn@cristinazn-GL553VD:~/Downloads/tslab/spatial$ ./bin/update_resources.sh 
cristinazn@cristinazn-GL553VD:~/Downloads/tslab/spatial$

mattfel1, 2019-11-13:
This looks like a problem with the libisl installation.  It seems like you managed to fix it though based on your other post.

closed 2019-11-14 by CristinaZN