stanford-ppl/spatial #283: many erros  when do bin/spatial HelloSpatial --sim

CristinaZN, 2019-11-13:
please help....
spatial-master$ bin/spatial HelloSpatial --sim
[info] Loading global plugins from /home/cristinazn/.sbt/1.0/plugins
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/cristinazn/Downloads/tslab/spatial-master/project
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/home/cristinazn/Downloads/tslab/spatial-master/)
[info] Set current project to apps (in build file:/home/cristinazn/Downloads/tslab/spatial-master/)
[info] Updating forge...
[info] Updating models...
[info] Updating poly...
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- edu.stanford.cs.dawn:models_2.12:1.1-SNAPSHOT      (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- org.scala-lang:scala-compiler:2.12.7               (depends on 1.0.6)
[warn] 	    +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- edu.stanford.cs.dawn:emul_2.12:1.1-SNAPSHOT        (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:poly_2.12:1.1-SNAPSHOT        (depends on 1.0.6)
[warn] 	    +- org.scala-lang:scala-compiler:2.12.7               (depends on 1.0.6)
[warn] 	    +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- edu.stanford.cs.dawn:forge_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- org.scala-lang:scala-compiler:2.12.7               (depends on 1.0.6)
[warn] 	    +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Updating argon...
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- edu.stanford.cs.dawn:forge_2.12:1.1-SNAPSHOT       (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:emul_2.12:1.1-SNAPSHOT        (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:argon_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- org.scala-lang:scala-compiler:2.12.7               (depends on 1.0.6)
[warn] 	    +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Updating spatial...
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- edu.stanford.cs.dawn:models_2.12:1.1-SNAPSHOT      (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:poly_2.12:1.1-SNAPSHOT        (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:argon_2.12:1.1-SNAPSHOT       (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:emul_2.12:1.1-SNAPSHOT        (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:spatial_2.12:1.1-SNAPSHOT     (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:forge_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- edu.stanford.cs.dawn:utils_2.12:1.1-SNAPSHOT       (depends on 1.0.6)
[warn] 	    +- org.scala-lang:scala-compiler:2.12.7               (depends on 1.0.6)
[warn] 	    +- org.scalatest:scalatest_2.12:3.0.5                 (depends on 1.0.6)
[warn] Run 'evicted' to see detailed eviction warnings
[info] Updating ...
[info] Done updating.
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn] 	* org.scala-lang.modules:scala-xml_2.12:1.1.0 is selected over 1.0.6
[warn] 	    +- edu.stanford.cs.dawn:models_2.12:1.1-SNAPSHOT      (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:poly_2.12:1.1-SNAPSHOT        (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:argon_2.12:1.1-SNAPSHOT       (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:emul_2.12:1.1-SNAPSHOT        (depends on 1.1.0)
[warn] 	    +- edu.stanford.cs.dawn:spatial_2.12:1.1-SNAPSHOT     (depends on 1.0.6)
[... 172 lines cut ...]
[error] 	at scala.reflect.internal.Types$TypeRef.seenFromOwnerInstantiated$1(Types.scala:2206)
[error] 	at scala.reflect.internal.Types$TypeRef.relativize(Types.scala:2210)
[error] 	at scala.reflect.internal.Types$TypeRef.$anonfun$baseTypeSeqImpl$2(Types.scala:2312)
[error] 	at scala.reflect.internal.Types$TypeRef$$Lambda$3761/308046723.apply(Unknown Source)
[error] 	at scala.reflect.internal.BaseTypeSeqs$BaseTypeSeq.map(BaseTypeSeqs.scala:145)
[error] 	at scala.reflect.internal.Types$TypeRef.baseTypeSeqImpl(Types.scala:2312)
[error] 	at scala.reflect.internal.Types.defineBaseTypeSeqOfTypeRef(Types.scala:2472)
[error] 	at scala.reflect.internal.Types.defineBaseTypeSeqOfTypeRef$(Types.scala:2463)
[error] 	at scala.reflect.internal.SymbolTable.defineBaseTypeSeqOfTypeRef(SymbolTable.scala:18)
[error] 	at scala.reflect.internal.Types$TypeRef.baseTypeSeq(Types.scala:2319)
[error] 	at scala.reflect.internal.BaseTypeSeqs.$anonfun$compoundBaseTypeSeq$1(BaseTypeSeqs.scala:193)
[error] 	at scala.reflect.internal.BaseTypeSeqs.compoundBaseTypeSeq(BaseTypeSeqs.scala:192)
[error] 	at scala.reflect.internal.BaseTypeSeqs.compoundBaseTypeSeq$(BaseTypeSeqs.scala:180)
[error] 	at scala.reflect.internal.SymbolTable.compoundBaseTypeSeq(SymbolTable.scala:18)
[error] 	at scala.reflect.internal.Types.defineBaseTypeSeqOfCompoundType(Types.scala:1505)
[error] 	at scala.reflect.internal.Types.defineBaseTypeSeqOfCompoundType$(Types.scala:1465)
[error] 	at scala.reflect.internal.SymbolTable.defineBaseTypeSeqOfCompoundType(SymbolTable.scala:18)
[error] 	at scala.reflect.internal.Types$CompoundType.baseTypeSeq(Types.scala:1373)
[error] 	at scala.reflect.internal.Types$PolyType.baseTypeSeq(Types.scala:2652)
[error] 	at scala.reflect.internal.Types$TypeRef.baseTypeSeqImpl(Types.scala:2306)
[error] 	at scala.reflect.internal.Types.defineBaseTypeSeqOfTypeRef(Types.scala:2472)
[error] 	at scala.reflect.internal.Types.defineBaseTypeSeqOfTypeRef$(Types.scala:2463)
[error] 	at scala.reflect.internal.SymbolTable.defineBaseTypeSeqOfTypeRef(SymbolTable.scala:18)
[error] 	at scala.reflect.internal.Types$TypeRef.baseTypeSeq(Types.scala:2319)
[error] 	at scala.reflect.internal.BaseTypeSeqs.$anonfun$compoundBaseTypeSeq$1(BaseTypeSeqs.scala:193)

mattfel1, 2019-11-13:
You should be able to fix this by increasing some of the Java options related to memory usage.  Unfortunately, Spatial has a relatively large memory footprint right now.  If you can get away with these values on your machine, can you try `export _JAVA_OPTIONS="-Xmx10g -Xss8912k -Xms4g"`?

CristinaZN, 2019-11-14:
ok, it works. but it report that "cannot compiling emptiness". 

**** SBT Command: ****
sbt "apps/runMain HelloSpatial --name HelloSpatial --sim"
**********************

Picked up _JAVA_OPTIONS: -Xmx10g -Xss8912k -Xms4g
[info] Loading global plugins from /home/cristinazn/.sbt/1.0/plugins
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/cristinazn/Downloads/tslab/spatial-master/project
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/home/cristinazn/Downloads/tslab/spatial-master/)
[info] Packaging /home/cristinazn/Downloads/tslab/spatial-master/utils/target/scala-2.12/utils_2.12-1.1-SNAPSHOT.jar ...
[info] Packaging /home/cristinazn/Downloads/tslab/spatial-master/forge/target/scala-2.12/forge_2.12-1.1-SNAPSHOT.jar ...
[info] Packaging /home/cristinazn/Downloads/tslab/spatial-master/poly/target/scala-2.12/poly_2.12-1.1-SNAPSHOT.jar ...
[info] Packaging /home/cristinazn/Downloads/tslab/spatial-master/models/target/scala-2.12/models_2.12-1.1-SNAPSHOT.jar ...
[info] Done packaging.
[info] Done packaging.
[info] Done packaging.
[info] Done packaging.
[info] Packaging /home/cristinazn/Downloads/tslab/spatial-master/argon/target/scala-2.12/argon_2.12-1.1-SNAPSHOT.jar ...
[info] Done packaging.
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Packaging /home/cristinazn/Downloads/tslab/spatial-master/apps/target/scala-2.12/apps_2.12-1.1-SNAPSHOT.jar ...
[info] Packaging /home/cristinazn/Downloads/tslab/spatial-master/target/scala-2.12/spatial_2.12-1.1-SNAPSHOT.jar ...
[info] Done packaging.
[info] Done packaging.
[info] Running HelloSpatial --name HelloSpatial --sim
[warn] No target specified. Specify target using: --fpga <device> or
[warn] override val target = <device>
[warn] Defaulting to 'Default' device.
[warn] Area model file Zynq_Area.csv for target Zynq was missing expected fields: 
[warn] BRAM
[info] Compiling HelloSpatial to /home/cristinazn/Downloads/tslab/spatial-master/gen/HelloSpatial/
[warn] libisl appears to be missing!  Please install http://isl.gforge.inria.fr/
[bug] An exception was encountered while compiling HelloSpatial: 
[bug]   Error compiling emptiness
[bug]   java.lang.Exception: Error compiling emptiness
[bug] This is due to a compiler bug. A log file has been created at: 
[bug]   /home/cristinazn/Downloads/tslab/spatial-master/logs/HelloSpatial//HelloSpatial_exception.log
[warn] 1 warning found
[failed] Total time: 4.9560 seconds

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error] 	at sbt.Run$.executeTrapExit(Run.scala:124)
[error] 	at sbt.Run.run(Run.scala:77)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error] 	at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.Try$.apply(Try.scala:209)
[error] 	at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (apps / Compile / runMain) Nonzero exit code: 1
[error] Total time: 48 s, completed Nov 14, 2019 6:56:05 PM
[error] Spatial compilation failed

and i tried `cd poly/resources && gcc emptiness.c` 
some files are not correctly included:

emptiness.c:6:10: fatal error: isl/vec.h: No such file or directory
 #include <isl/vec.h>
          ^~~~~~~~~~~
compilation terminated.

how to figure it out

mattfel1, 2019-11-14:
After installing libisl-dev, if you go to poly/resources and try compiling with the -lisl flag included, do you still get the error about isl/vec.h missing?  `g++ emptiness.c -lisl`

If so, I'm curious why it can't find the isl headers.  If you do `\`g++ -print-prog-name=cc1plus\` -v` can you check if the isl is installed into one of those locations? For example in my installation, it finds the isl headers in `/usr/include/isl/*.h`

Maybe you can add the correct path to `CPLUS_INCLUDE_PATH` env variable?

CristinaZN, 2019-11-15:
there is no errors about isl/vec.h missing any more. 

g++ emptiness.c -lisl
/resources$ g++ emptiness.c 
/tmp/ccBGj1dV.o: In function `main':
emptiness.c:(.text+0x6b): undefined reference to 'isl_ctx_alloc'
emptiness.c:(.text+0x99): undefined reference to 'isl_basic_set_read_from_file'
emptiness.c:(.text+0xa9): undefined reference to 'isl_basic_set_plain_is_empty'
emptiness.c:(.text+0xca): undefined reference to 'isl_basic_set_is_empty'
emptiness.c:(.text+0xf5): undefined reference to 'isl_basic_set_free'
collect2: error: ld returned 1 exit status
i thinks these error can be ignored, though i'm curious about why there are these undefined references.

closed 2019-11-15 by CristinaZN