stanford-ppl/spatial #307: Exception compiling HelloSpatial (isl dependency) on AWS F1

feio-ai, 2020-06-09:
I'm attempting to run the example app on EC2 instance and encountering the below problem. I've followed the aws instructions and think the issue is with the isl lib.

When executing command:
bin/spatial HelloSpatial --synth --fpga=AWS_F1 --vv

I receive the below exception:

``` bash
**** SBT Command: ****
sbt "apps/runMain HelloSpatial --name HelloSpatial --synth --fpga=AWS_F1 --vv"
**********************

Picked up _JAVA_OPTIONS: -Xms1024m -Xss256m -Xmx16g -XX:MaxMetaspaceSize=16g
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/centos/spatial-multiverse/spatial/project
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/home/centos/spatial-multiverse/spatial/)
[info] Packaging /home/centos/spatial-multiverse/spatial/poly/target/scala-2.12/poly_2.12-1.1-SNAPSHOT.jar ...
[info] Done packaging.
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Packaging /home/centos/spatial-multiverse/spatial/target/scala-2.12/spatial_2.12-1.1-SNAPSHOT.jar ...
[info] Done packaging.
[info] Running HelloSpatial --name HelloSpatial --synth --fpga=AWS_F1 --vv
[warn] Area model file AWS_F1_Area.csv for target AWS_F1 was missing expected fields: 
[warn] BRAM
[info] Compiling HelloSpatial to /home/centos/spatial-multiverse/spatial/gen/HelloSpatial/
[info] Logging HelloSpatial to /home/centos/spatial-multiverse/spatial/logs/HelloSpatial/
[bug] An exception was encountered while compiling HelloSpatial: 
[bug]   Error compiling emptiness
[bug]   java.lang.Exception: Error compiling emptiness
[bug]   poly.ISL.liftedTree5$1(ISL.scala:105)
[bug]   poly.ISL.poly$ISL$$proc(ISL.scala:74)
[bug]   poly.ISL.poly$ISL$$proc$(ISL.scala:14)
[bug]   spatial.Spatial$SpatialISL.poly$ISL$$proc$lzycompute(Spatial.scala:37)
[bug]   spatial.Spatial$SpatialISL.poly$ISL$$proc(Spatial.scala:37)
[bug]   poly.ISL.init(ISL.scala:116)
[bug]   poly.ISL.startup(ISL.scala:120)
[bug]   poly.ISL.startup$(ISL.scala:120)
[bug]   spatial.Spatial$SpatialISL.startup(Spatial.scala:37)
[bug]   spatial.Spatial.runPasses(Spatial.scala:58)
[bug]   spatial.Spatial.runPasses$(Spatial.scala:56)
[bug]   HelloSpatial$.runPasses(HelloSpatial.scala:3)
[bug]   argon.Compiler.$anonfun$compileProgram$1(Compiler.scala:129)
[bug]   scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[bug]   utils.Instrument.apply(Instrument.scala:29)
[bug]   argon.Compiler.compileProgram(Compiler.scala:119)
[bug]   argon.Compiler.compileProgram$(Compiler.scala:119)
[bug]   HelloSpatial$.compileProgram(HelloSpatial.scala:3)
[bug]   argon.Compiler.$anonfun$execute$1(Compiler.scala:198)
[bug]   scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[bug]  .. [see /home/centos/spatial-multiverse/spatial/logs/HelloSpatial/HelloSpatial_exception.log]
[bug] This is due to a compiler bug. A log file has been created at: 
[bug]   /home/centos/spatial-multiverse/spatial/logs/HelloSpatial//HelloSpatial_exception.log
[failed] Total time: 0.9450 seconds

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error] 	at sbt.Run$.executeTrapExit(Run.scala:124)
[error] 	at sbt.Run.run(Run.scala:77)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error] 	at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.Try$.apply(Try.scala:209)
[error] 	at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (apps / Compile / runMain) Nonzero exit code: 1
[error] Total time: 7 s, completed Jun 9, 2020 11:10:22 PM
[error] Spatial compilation failed
```

When compiling emptiness.c without flag -lisl and get (with flag no exception raised):

``` bash
/tmp/cczE5ftS.o: In function `main':
emptiness.c:(.text+0x66): undefined reference to `isl_ctx_alloc'
emptiness.c:(.text+0x94): undefined reference to `isl_basic_set_read_from_file'
emptiness.c:(.text+0xa4): undefined reference to `isl_basic_set_plain_is_empty'
emptiness.c:(.text+0xc0): undefined reference to `isl_basic_set_is_empty'
emptiness.c:(.text+0xe6): undefined reference to `isl_basic_set_free'
collect2: error: ld returned 1 exit status
```

Finally, `g++ -print-prog-name=cc1plus` -v results in: 
``` bash
ignoring nonexistent directory "/usr/lib/gcc/x86_64-redhat-linux/4.8.5/include-fixed"
ignoring nonexistent directory "/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../x86_64-redhat-linux/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../include/c++/4.8.5
 /usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../include/c++/4.8.5/x86_64-redhat-linux
 /usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../include/c++/4.8.5/backward
 /usr/lib/gcc/x86_64-redhat-linux/4.8.5/include
 /usr/local/include
 /usr/include
End of search list.
```

I'm sure this relates to the ISL lib but can't figure where I've gone wrong. Thanks!

mattfel1, 2020-06-10:
Hi.  Can you confirm you have libisl installed and check what it says if you try to compile emptiness.c with the -lisl flag?  `gcc emptiness.c -lisl -o emptiness` 

If this works, the really quick solution is to just move the emptiness bin to somewhere on your path (Spatial will have attempted to place it in ~/bin and add that to your path, but anywhere else on your path will work when you "install" it manually).

feio-ai, 2020-06-10:
Perfect thank you that worked. After moving the emptiness bin I had to move the libisl.so to /lib64