stanford-ppl/spatial #309: Categorical HM integration with Spatial kernel implementations

mattfel1, 2020-06-15:
For example, can we do something like:

```
def innerProdMatMult(tileA: SRAM2[T], tileB: SRAM2[T], tileC: SRAM2[T]): Void = {
  Foreach(...){...} // fill tileC using inner products
}

def outerProdMatMult(tileA: SRAM2[T], tileB: SRAM2[T], tileC: SRAM2[T]): Void = {
  Foreach(...){...} // fill tileC using outer products
}

Accel{ 
  ...
  innerProductMatMult(tileA, tileB, tileC) (innerProductMatMult, outerProductMatMult)
  ...
} 
```

mattfel1, 2020-06-15:
This may be problematic with Spatial's staging.  Maybe you can use blackboxes as containers for kernel switching since those won't get dropped.  Maybe there can be a generic BlackBoxUse node and store the possible drop-in replacements as Parameter metadata

David-Durst, 2020-06-15:
To add some more motivation: Aetherling (https://github.com/David-Durst/embeddedHaskellAetherling) has multiple options for the same operation with different performance-resources trade-offs. This feature would allow treating both options as a single operator that can be configured by HM,