stanford-ppl/spatial #311: Unable to compile HelloSpatial when run "bin/spatial HelloSpatial --synth --fpga=ZCU"

yrq0208, 2020-07-11:
Hi all,

I have encountered this "java.util.concurrent.ExecutionException: java.lang.OutOfMemoryError: GC overhead limit exceeded" error when I tried to compile HelloSpatial using Spatial. This seems to be a known issue? I have tried using "export _JAVA_OPTIONS="-Xmx10g -Xss8912k -Xms4g" " but it doesn't seem to solve the problem. Some more details are as follows,

[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/ruiqi/spatial/project
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/home/ruiqi/spatial/)
[info] Set current project to apps (in build file:/home/ruiqi/spatial/)
[success] Total time: 11 s, completed 11-Jul-2020 12:06:35
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/ruiqi/spatial/project
[info] Loading settings from build.sbt ...
[info] Set current project to spatial (in build file:/home/ruiqi/spatial/)
[info] Set current project to test (in build file:/home/ruiqi/spatial/)
[info] Compiling 272 Scala sources to /home/ruiqi/spatial/test/target/scala-2.12/test-classes ...
[info] /home/ruiqi/spatial/test/spatial/tests/feature/math/FixBasics2.scala:69:9: Type annotation recommended for var declaration.
[info]     var allgood = true
[info]         ^
[info] /home/ruiqi/spatial/test/spatial/tests/feature/math/FloatBasics2.scala:70:9: Type annotation recommended for var declaration.
[info]     var allgood = true
[info]         ^
[warn] /home/ruiqi/spatial/test/spatial/tests/feature/streaming/FilterStream1D.scala:49:97: postfix operator !! should be enabled
[warn] by making the implicit value scala.language.postfixOps visible.
[warn] This can be achieved by adding the import clause 'import scala.language.postfixOps'
[warn] or by setting the compiler option -language:postfixOps.
[warn] See the Scaladoc for value scala.language.postfixOps for a discussion
[warn] why the feature should be explicitly enabled.
[warn]     val filename = "cat ../src/findedges.py | grep \"calibfile =\" | sed \"s/calibfile = //g\"" !!
[warn]                                                                                                 ^
[warn] /home/ruiqi/spatial/test/spatial/tests/feature/streaming/FilterStream1D.scala:52:35: postfix operator ! should be enabled
[warn] by making the implicit value scala.language.postfixOps visible.
[warn]     "python3 ../src/findedges.py" !
[warn]                                   ^
[warn] /home/ruiqi/spatial/test/spatial/tests/feature/streaming/FilterStream1D.scala:57:43: postfix operator ! should be enabled
[warn] by making the implicit value scala.language.postfixOps visible.
[warn]     s"echo EXAMPLE SCALA METAPROGRAMMING" !
[warn]                                           ^
[error] ## Exception when compiling 272 sources to /home/ruiqi/spatial/test/target/scala-2.12/test-classes
[error] GC overhead limit exceeded
[error] scala.reflect.internal.Symbols$Symbol.info_$eq(Symbols.scala:1550)
[error] scala.reflect.internal.Symbols$Symbol.setInfo(Symbols.scala:1556)
[error] scala.reflect.internal.Symbols$Symbol.cloneSymbol(Symbols.scala:2007)
[error] scala.reflect.internal.Symbols$Symbol.cloneSymbol(Symbols.scala:2002)
[error] scala.reflect.internal.Symbols$Symbol.cloneSymbol(Symbols.scala:2000)
[error] scala.reflect.internal.Symbols$Symbol.cloneSymbol(Symbols.scala:1996)
[error] scala.reflect.internal.Symbols.$anonfun$cloneSymbols$1(Symbols.scala:3684)
[error] scala.reflect.internal.Symbols$$Lambda$3678/171087126.apply(Unknown Source)
[error] scala.reflect.internal.Symbols.deriveSymbols(Symbols.scala:3623)
[error] scala.reflect.internal.Symbols.deriveSymbols$(Symbols.scala:3622)
[error] scala.reflect.internal.SymbolTable.deriveSymbols(SymbolTable.scala:18)
[error] scala.reflect.internal.Symbols.cloneSymbols(Symbols.scala:3684)
[error] scala.reflect.internal.Symbols.cloneSymbols$(Symbols.scala:3683)
[error] scala.reflect.internal.SymbolTable.cloneSymbols(SymbolTable.scala:18)
[error] scala.reflect.internal.tpe.TypeMaps$TypeMap.mapOver(TypeMaps.scala:247)
[error] scala.reflect.internal.tpe.TypeMaps$TypeMap.$anonfun$mapOver$1(TypeMaps.scala:125)
[error] scala.reflect.internal.tpe.TypeMaps$TypeMap.mapOver(TypeMaps.scala:199)
[error] scala.reflect.internal.transform.UnCurry$$anon$1.apply(UnCurry.scala:53)
[error] scala.tools.nsc.transform.UnCurry$UnCurryTransformer.mainTransform(UnCurry.scala:525)
[error] scala.tools.nsc.transform.UnCurry$UnCurryTransformer.transform(UnCurry.scala:111)
[error] scala.tools.nsc.transform.UnCurry$UnCurryTransformer.$anonfun$mainTransform$12(UnCurry.scala:476)
[error] scala.tools.nsc.transform.UnCurry$UnCurryTransformer.mainTransform(UnCurry.scala:394)
[error] scala.tools.nsc.transform.UnCurry$UnCurryTransformer.transform(UnCurry.scala:111)
[error] scala.tools.nsc.transform.UnCurry$UnCurryTransformer.transform(UnCurry.scala:71)
[error] scala.reflect.api.Trees$Transformer.$anonfun$transformStats$1(Trees.scala:2589)
[error] scala.reflect.api.Trees$Transformer.transformStats(Trees.scala:2587)
[error] scala.reflect.internal.Trees.itransform(Trees.scala:1392)
[error] scala.reflect.internal.Trees.itransform$(Trees.scala:1362)
[error] scala.reflect.internal.SymbolTable.itransform(SymbolTable.scala:18)
[error] scala.reflect.internal.SymbolTable.itransform(SymbolTable.scala:18)
[... 60 lines cut ...]
[error] 	at scala.reflect.internal.SymbolTable.cloneSymbols(SymbolTable.scala:18)
[error] 	at scala.reflect.internal.tpe.TypeMaps$TypeMap.mapOver(TypeMaps.scala:247)
[error] 	at scala.reflect.internal.tpe.TypeMaps$TypeMap.$anonfun$mapOver$1(TypeMaps.scala:125)
[error] 	at scala.reflect.internal.tpe.TypeMaps$TypeMap.mapOver(TypeMaps.scala:199)
[error] 	at scala.reflect.internal.transform.UnCurry$$anon$1.apply(UnCurry.scala:53)
[error] 	at scala.tools.nsc.transform.UnCurry$UnCurryTransformer.mainTransform(UnCurry.scala:525)
[error] 	at scala.tools.nsc.transform.UnCurry$UnCurryTransformer.transform(UnCurry.scala:111)
[error] 	at scala.tools.nsc.transform.UnCurry$UnCurryTransformer.$anonfun$mainTransform$12(UnCurry.scala:476)
[error] 	at scala.tools.nsc.transform.UnCurry$UnCurryTransformer.mainTransform(UnCurry.scala:394)
[error] 	at scala.tools.nsc.transform.UnCurry$UnCurryTransformer.transform(UnCurry.scala:111)
[error] 	at scala.tools.nsc.transform.UnCurry$UnCurryTransformer.transform(UnCurry.scala:71)
[error] 	at scala.reflect.api.Trees$Transformer.$anonfun$transformStats$1(Trees.scala:2589)
[error] 	at scala.reflect.api.Trees$Transformer.transformStats(Trees.scala:2587)
[error] 	at scala.reflect.internal.Trees.itransform(Trees.scala:1392)
[error] 	at scala.reflect.internal.Trees.itransform$(Trees.scala:1362)
[error] 	at scala.reflect.internal.SymbolTable.itransform(SymbolTable.scala:18)
[error] 	at scala.reflect.internal.SymbolTable.itransform(SymbolTable.scala:18)
[error] 	at scala.reflect.api.Trees$Transformer.transform(Trees.scala:2555)
[error] 	at scala.tools.nsc.transform.TypingTransformers$TypingTransformer.transform(TypingTransformers.scala:44)
[error] java.util.concurrent.ExecutionException: java.lang.OutOfMemoryError: GC overhead limit exceeded
[error] Use 'last' for the full log.

I wonder has anyone has encountered this problem before?

Thanks in advance.

mattfel1, 2020-07-12:
There are two things you can try:

1) Bump up the _JAVA_OPTIONS parameters even further and confirm that they are getting picked up when you call the compilation.
2) Just remove everything in test/spatial/tests/*.  This crash is happening when it tries to compile all of our hundreds of unit tests at once.  You probably don't need all of them so deleting them should have a much lower memory footprint when compiling for the first time.

yrq0208, 2020-07-16:
Hi @mattfel1, thanks for the reply. The problem is solved by using larger parameters of _JAVA_OPTIONS.

closed 2020-07-18 by yrq0208